Sorunome/arduino-upload #1: Question About 'Serial Monitor'

Rperes, 2016-04-30:
Question:
Does  'serial monitor' work just to report activity from arduino?

I can see all text received from arduino, well formatted but i can't send commands from keyboard to arduino  although see represented on the 'serial Monitor screen! !!!

('arduino-upload packet'  seems work nice and stable with my iMac  OS X Ver 10.11.4.
Tks,
Jose

Sorunome, 2016-04-30:
I have not implemented sending anything, any idea how it should work UI-wise?

Rperes, 2016-04-30:
Thanks for your info.
It will be great to use that functionality! 
Rgds.
Jose.

Sorunome, 2016-04-30:
Do you have an idea how it should look like GUI-wise?
An input box at the bottom of the screen or something?

Rperes, 2016-04-30:
If you want to give a profissional look, the best ideia for me is . . .  look like the  "Serial Monitor" for the Shell, just with a icon to open/close the working window.

Sorunome, 2016-04-30:
Added in https://github.com/Sorunome/arduino-upload/releases/tag/v0.6.0

closed 2016-04-30 by Sorunome