Sorunome/arduino-upload #2: Cannot install

chdsbd, 2016-06-02:
When I attempt to install the package, I get the following error.

```
> serialport@4.0.0-beta3 install /private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport
> node-pre-gyp install --fallback-to-build

Unsupported target version: 0.36.8

npm WARN installMany node-pre-gyp was bundled with serialport@4.0.0-beta3, but bundled package wasn't found in unpacked tree
npm WARN engine deep-extend@0.4.1: wanted: {"node":">=0.12.0","iojs":">=1.0.0"} (current: {"node":"0.10.40","npm":"2.13.3"})
node-pre-gyp ERR! install error
node-pre-gyp ERR! stack Error: Unsupported target version: 0.36.8
node-pre-gyp ERR! stack     at get_runtime_abi (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "node" "/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/node_modules/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok
npm ERR! Darwin 14.5.0
npm ERR! argv "/opt/homebrew-cask/Caskroom/atom/1.7.3/Atom.app/Contents/Resources/app/apm/bin/node" "/opt/homebrew-cask/Caskroom/atom/1.7.3/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/chris/.atom/.apm/.apmrc" "--userconfig" "/Users/chris/.atom/.apmrc" "install" "/private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/d-11651-22381-1olqfvs/package.tgz" "--target=0.36.8" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.0-beta3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.0-beta3 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/dx/ml7z6ck95dn4rx_vn1t3f6mh0000gn/T/apm-install-dir-11651-22381-dza0i/npm-debug.log
```

Sorunome, 2016-06-03:
Unfortunately this is probably because of the npm version packed in apm is out-of-date.  
However you can still install by manually downloading the folder, running `npm install serialport` and then `apm install`

chdsbd, 2016-06-03:
I tried installing using the method you specified, but it produced a similar error.
```
> serialport@4.0.0-beta3 install /Users/chris/Downloads/arduino-upload/node_modules/serialport
> node-pre-gyp install --fallback-to-build

Unsupported target version: 0.36.8

npm WARN installMany node-pre-gyp was bundled with serialport@4.0.0-beta3, but bundled package wasn't found in unpacked tree
node-pre-gyp ERR! install error
node-pre-gyp ERR! stack Error: Unsupported target version: 0.36.8
node-pre-gyp ERR! stack     at get_runtime_abi (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/lib/util/versioning.js:156:23)
node-pre-gyp ERR! stack     at Object.module.exports.evaluate (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/lib/util/versioning.js:277:19)
node-pre-gyp ERR! stack     at install (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/lib/install.js:148:31)
node-pre-gyp ERR! stack     at Object.self.commands.(anonymous function) [as install] (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/lib/node-pre-gyp.js:48:37)
node-pre-gyp ERR! stack     at run (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/bin/node-pre-gyp:79:30)
node-pre-gyp ERR! stack     at Object.<anonymous> (/Users/chris/Downloads/arduino-upload/node_modules/node-pre-gyp/bin/node-pre-gyp:131:1)
node-pre-gyp ERR! stack     at Module._compile (module.js:456:26)
node-pre-gyp ERR! stack     at Object.Module._extensions..js (module.js:474:10)
node-pre-gyp ERR! stack     at Module.load (module.js:356:32)
node-pre-gyp ERR! stack     at Function.Module._load (module.js:312:12)
node-pre-gyp ERR! System Darwin 14.5.0
node-pre-gyp ERR! command "node" "/Users/chris/Downloads/arduino-upload/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /Users/chris/Downloads/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v0.10.40
node-pre-gyp ERR! node-pre-gyp -v v0.6.28
node-pre-gyp ERR! not ok
npm ERR! Darwin 14.5.0
npm ERR! argv "/opt/homebrew-cask/Caskroom/atom/1.7.3/Atom.app/Contents/Resources/app/apm/bin/node" "/opt/homebrew-cask/Caskroom/atom/1.7.3/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/chris/.atom/.apm/.apmrc" "--userconfig" "/Users/chris/.atom/.apmrc" "install" "--target=0.36.8" "--arch=x64"
npm ERR! node v0.10.40
npm ERR! npm  v2.13.3
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.0-beta3 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.0-beta3 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! This is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/chris/Downloads/arduino-upload/npm-debug.log
```

Sorunome, 2016-06-04:
did you pay attention to use `npm` instead of `apm` for installing serialport?  
If so, what do the following commands give you?  
`npm --version`  
`apm --version`

chdsbd, 2016-06-05:
I did follow the instructions. Here is the output of those commands.
```
$ apm --version
apm  1.9.2
npm  2.13.3
node 0.10.40
python 3.5.1
git 2.8.1
```

```
$ npm --version
3.9.3
```

Sorunome, 2016-06-05:
Try updating your npm, it's quite heavily out-of-date.

Sorunome, 2016-06-05:
This is odd, in the output you gave me for the serial port installation it says `npm ERR! npm  v2.13.3` which would point to the npm thing packed in apm being used.... o.O

simonvanderveldt, 2016-06-12:
@Sorunome AFAIK it will still fail because `apm install` will trigger the [install script](https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/master/package.json#L93) from node-serialport which uses `node-pre-gyp` which doesn't work in Atom/Electron, see https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/538#issuecomment-115207973

Sorunome, 2016-06-12:
@simonvanderveldt nope it won't fail, as npm installs serialport and when running `apm install` it'll notice that serialport is already present and won't install it, thus preventing node-pre-gyp issues. At least that worked for me, but I guess `npm install` would work, too

simonvanderveldt, 2016-06-12:
@Sorunome As a clarification to my previous message, the AFAIK relates to what I think is the cause, the installation does fail.

`npm install` didn't work in the end, so I've switched to just calling the correct shell commands instead of using this package :)

Sorunome, 2016-06-12:
@simonvanderveldt you should try `npm install serialport` and then `apm install`, that is what I do to install this package

simonvanderveldt, 2016-06-12:
@Sorunome yeah, that doesn't work and gives the error @chdsbd posted.
I guess it's caused by what I described in https://github.com/Sorunome/arduino-upload/issues/2#issuecomment-225430253

I'm on Atom 1.8.0 and have tried with both npm 2 and 3.

Sorunome, 2016-06-12:
Well, that error shows that `npm install serialport` somehow uses npm 2.13.3 which is the one packaged into apm, the `npm` version you have installed should be more like 3.9.3, so maybe make absolutely sure you use that one by running like `/usr/bin/npm install serialport` ?

simonvanderveldt, 2016-06-12:
To clarify: `npm install serialport` from within `~/.atom/packages/arduino-upload` works, if you then run `apm install` from within `~/.atom/packages/arduino-upload` it will fail with the above message, probably because of the install script within serialport.

Sorunome, 2016-06-12:
Hm, weired, to me that is working. Perhaps try removing the serialport dependency in the `package.json`? Also, which distro are you running (so that I can try to reproduce)?

simonvanderveldt, 2016-06-12:
This is on OS X (10.11.4 to be exact).

Sorunome, 2016-06-14:
@simonvanderveldt @chdsbd So the reson that error appeared was because, for whatever reason, i had put the git version of serialport in package.json. I fixed it now by requiring >=2.1.0 like I used to, so `npm install serialport` followed by `apm install` is working again. Please use this repo as I didn't upload the changes via apm yet.

Sorunome, 2016-06-15:
I disabled now the serialport code until the package actually works under atom, since atom 1.8.0 it failed to load because of serialport :(

At least you can still verify / build / upload!

Sorunome, 2016-09-07:
Fixed with https://github.com/Sorunome/arduino-upload/releases/tag/v1.0.0

Requires atom 1.10 or later

closed 2016-09-07 by Sorunome

jango-fx, 2016-09-07:
i still get

`apm install arduino-upload
Installing arduino-upload to /Users/USER/.atom/packages ✗

> serialport@4.0.1 install /private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport
> node-pre-gyp install --fallback-to-build

Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp build --fallback-to-build --module=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0
└── (empty)

node-pre-gyp ERR! Tried to download: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.1/serialport-v4.0.1-electron-v0.37-darwin-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.1 and electron@0.37.8 (electron-v0.37 ABI) (falling back to source compile with node-gyp) 
make: *** No rule to make target `../../../../../../../../../../../../Users/USER/Dropbox/1_cfgs', needed by `Makefile'.  Stop.
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Darwin 15.6.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/.bin/node-gyp" "build" "--fallback-to-build" "--module=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release"
gyp ERR! cwd /private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp build --fallback-to-build --module=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 15.6.0
node-pre-gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/node_modules/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.29
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/package.json'
npm WARN apm-install-dir-11687-42524-16358x0 No description
npm WARN apm-install-dir-11687-42524-16358x0 No repository field.
npm WARN apm-install-dir-11687-42524-16358x0 No README data
npm WARN apm-install-dir-11687-42524-16358x0 No license field.
npm ERR! Darwin 15.6.0
npm ERR! argv "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/USER/.atom/.apm/.apmrc" "--userconfig" "/Users/USER/.atom/.apmrc" "install" "/private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/d-11687-42524-nyckbj/package.tgz" "--runtime=electron" "--target=0.37.8" "--arch=x64" "--global-style"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE

[... cut ...]
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/8z/js7_8g7s3mg1y92n3vystfkw0000gn/T/apm-install-dir-11687-42524-16358x0/npm-debug.log
npm ERR! code 1`

`apm --version
apm  1.12.5
npm  3.10.5
node 4.4.5
python 2.7.10
git 2.7.4`

`npm --version
3.10.3
`
Atom v1.10.1

Sorunome, 2016-09-07:
which atom version are you at? what do you get with `apm --version` ?

jango-fx, 2016-09-07:
see above. 

btw, serialport was installed via npm:
```
npm install serialport

> serialport@4.0.1 install /Users/USERNAME/node_modules/serialport
> node-pre-gyp install --fallback-to-build

[serialport] Success: "/Users/USERNAME/node_modules/serialport/build/Release/serialport.node" is installed via remote
/Users/USERNAME
└── serialport@4.0.1 
```

Sorunome, 2016-09-08:
if the `apm --version` output didn't change that means you are not on atom 1.10, they packaged a newer version of node with it (i think). The entire package should install via apm now, try upgrading stom, removing the node_modules folder and running `apm install`