Sorunome/arduino-upload #3: Error: "File isn't part of an Arduino sketch!"

luisbaldissera, 2016-06-28:
Using any of the commands the error "File isn't part of an Arduino sketch!" appears even when it is working on a .ino file inside of its respective sketch folder with the same file name

Sorunome, 2016-06-28:
Are you triggering the build process via the command palette or the drop-down menu?

luisbaldissera, 2016-06-28:
command pallete

Sorunome, 2016-06-28:
hm, which OS are you running & which atom version?

luisbaldissera, 2016-06-28:
but I tried via drop-down menu and I got the same error

luisbaldissera, 2016-06-28:
Sorry, I'm begginer but how I see the Atom version? hehe
I'm using Windows 10

luisbaldissera, 2016-06-28:
In my local Atom folder, there are 3 subfolders: app-1.7.4, app-1.7.3 and app-1.8.0... I suppose I'm using 1.8.0

luisbaldissera, 2016-06-28:
Ah, yes... now I checked, I'm really using Atom 1.8.0

Sorunome, 2016-06-28:
Which version of the arduino IDE do you have installed?

luisbaldissera, 2016-06-28:
Arduino 1.6.9

Sorunome, 2016-06-28:
Fixed in https://github.com/Sorunome/arduino-upload/commit/20820844c949b4c441ba2381e1974f26f8d6ff44 just update your package

closed 2016-06-28 by Sorunome

hex705, 2018-01-12:
This bug seems to be back -- or is for me.
When trying to upload I get the FILE isn't part of sketch error.
The FILE is, really.

All packages updated.
OSX 10.10.5 Yosemite
ATOM 1.23.3
Package 1.3.2
PATH = /Applications/Arduino.app  (note .app required on my system).
Error from command line as well as drop down.