Sorunome/arduino-upload #4: Uncaught Error: spawn EACCES

semsturgut, 2016-07-08:
[Enter steps to reproduce below:]

1. ...
2. ...

**Atom Version**: 1.8.0
**System**: Ubuntu 16.04
**Thrown From**: [arduino-upload](https://github.com/Sorunome/arduino-upload) package, v0.7.1

### Stack Trace

Uncaught Error: spawn EACCES

```
At internal/child_process.js:298

Error: spawn EACCES
    at exports._errnoException (util.js:856:11)
    at ChildProcess.spawn (internal/child_process.js:298:11)
    at exports.spawn (child_process.js:374:9)
    at Object.module.exports.ArduinoUpload.build (/home/sems/.atom/packages/arduino-upload/lib/arduino-upload.coffee:103:16)
    at atom-workspace.subscriptions.add.atom.commands.add.arduino-upload:verify (/home/sems/.atom/packages/arduino-upload/lib/arduino-upload.coffee:64:33)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/usr/share/atom/resources/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/usr/share/atom/resources/app.asar/src/atom-environment.js:1100:28)
    at EventEmitter.outerCallback (/usr/share/atom/resources/app.asar/src/application-delegate.js:327:25)
    at emitTwo (events.js:87:13)
    at EventEmitter.emit (events.js:172:7)
```

### Commands

```
     -3:01.5.0 arduino-upload:verify (ol.tree-view.full-menu.list-tree.has-collapsable-children.focusable-panel)
     -2:34.5.0 core:save (atom-text-editor.editor.is-focused)
     -2:10.6.0 command-palette:toggle (atom-text-editor.editor.is-focused)
     -2:07.0 core:move-down (atom-text-editor.editor.mini.is-focused)
     -2:06.7.0 core:confirm (atom-text-editor.editor.mini.is-focused)
     -2:06.7.0 settings-view:view-installed-packages (atom-text-editor.editor)
     -1:24.9.0 editor:delete-to-beginning-of-word (atom-text-editor#arduino-upload.arduinoExecutablePath.editor.mini.is-focused)
  9x -1:23.7.0 core:backspace (atom-text-editor#arduino-upload.arduinoExecutablePath.editor.mini.is-focused)
     -1:11.2.0 core:confirm (atom-text-editor#arduino-upload.arduinoExecutablePath.editor.mini.is-focused)
     -0:00.7.0 arduino-upload:verify (atom-text-editor.editor.is-focused)
```

### Config

```json
{
  "core": {
    "disabledPackages": [
      "eclipse-keybindings"
    ]
  },
  "arduino-upload": {
    "arduinoExecutablePath": "/home/sems/arduino-1.6.8"
  }
}
```

### Installed Packages

```coffee
# User
arduino-upload, v0.7.1
atom-beautify, v0.29.9
autocomplete-clang, v0.9.4
autocomplete-plus, v2.31.1
autocomplete-python, v1.8.1
language-arduino, v0.4.1
linter, v1.11.14
linter-clang, v3.4.4
linter-flake8, v1.13.2
minimap, v4.24.7
python-tools, v0.6.8

# Dev
No dev packages
```

Sorunome, 2016-07-08:
Do you have the arduino ide installed? It uses it in the background. If so, which version?

ay13, 2016-07-09:
I am having the same issue.

Arduino IDE 1.7.10
Mac El Capitan (10.11.5)
Java 8 update 91
Atom 1.8.0

Sorunome, 2016-07-10:
what happens if you run `arduino` or whatever you set as arduinoExecutablePath manually in the terminal? Perhaps you need execution permissions?

Sorunome, 2016-07-12:
@semsturgut did your issue get resolved? If so, how?

chipmc, 2016-09-13:
I am having the same issue.  I had to edit the path in your plug-in settings to: /Applications/Arduino.app/Contents/MacOS

I have also added this to the $PATH.  If I open a terminal window and type "arduino" the IDE launches.  Any ideas?

Sorunome, 2016-09-13:
Did you try editing in the arduino-upload settings the "Arduino Executable Path" to the full path to the executable?

chipmc, 2016-09-13:
yes, I was getting the "cannot spawn Arduino" error until I edited the settings to be " /Applications/Arduino.app/Contents/MacOS" and then I got the error message on this thread.

Sorunome, 2016-09-13:
Unfortunately I don't have any OSX device to debug this :( Arduino is invoced here: https://github.com/Sorunome/arduino-upload/blob/master/lib/arduino-upload.coffee#L103

chipmc, 2016-09-13:
Oh, right.  Added the executable itself and it works!  Thank you!

taldridge, 2017-02-23:
Error still not resolved.  Is there any support for this?
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.14.3 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.12.3
**Thrown From**: [arduino-upload](https://github.com/Sorunome/arduino-upload) package 1.0.3

### Stack Trace

Uncaught Error: spawn EACCES

```
At internal/child_process.js:313

Error: spawn EACCES
    at exports._errnoException (util.js:1026:11)
    at ChildProcess.spawn (internal/child_process.js:313:11)
    at exports.spawn (child_process.js:392:9)
    at Object.build (/packages/arduino-upload/lib/arduino-upload.coffee:103:16)
    at /packages/arduino-upload/lib/arduino-upload.coffee:64:33)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/app.asar/src/atom-environment.js:1164:28)
    at EventEmitter.outerCallback (/app.asar/src/application-delegate.js:320:25)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
```

### Non-Core Packages

```
arduino-upload 1.0.3 
language-arduino 0.4.2 
```

Sorunome, 2017-02-23:
Please open a new issue next time, don't try to resurrect old closed issues.

Also, this seems like a duplicate of #13 by any chance?

taldridge, 2017-02-23:
Sorry, new user and I need this to work.

Sorunome, 2017-02-23:
allright, i'll get to it eventaulyl then ^.^

Sorunome, 2017-02-23:
As I don't have a mac to actually test it on i'll try to fix #13 and if that doesn't fix it, i'll need more information.

However, I'm currently rather busy IRL so it can take some time...

Sorunome, 2017-02-23:
I actually confused this with another issue and it's better to have this open for now, woops, my bad

varung97, 2017-03-01:
I'm a Mac user. The fix that worked for me was:
<br>
### config.cson
```
"arduino-upload":
    arduinoExecutablePath: "/Applications/Arduino.app/Contents/MacOS/Arduino"
```

Sorunome, 2017-03-02:
You are able to change that setting without actually editing the config manually, that is that `arduino executable path`-option

taldridge, 2017-03-03:
Unfortunately, none of this worked for me on my mac.

Sorunome, 2017-03-26:
I don#t have a mac to test, any help would be appreciated. I suspect it being something with the path stuff being off, if somebody could do some investigation it would be awesome.

Sorunome, 2017-09-01:
I added default paths for macos and windows recently. https://github.com/Sorunome/arduino-upload/blob/f8f3586385c6de24b93b941214820bd02ef2ac82/lib/util.coffee line 5 to 13

closed 2017-09-01 by Sorunome

welrbraga, 2018-01-04:
Same here (but solved). I think that the problem is that the instruction is not clear. The "Arduino Executable Path" field is to inform the executable file with its path, not only the path. 

eg. My Arduino IDE is instaled in **/home/wbraga/arduino-1.8.5**. If I inform only it I see the same error message about "Uncaught Error spawn EACCESS". The executable file for Arduino is  **/home/wbraga/arduino-1.8.5/arduino**. That is exactly what I have to write to not see error and everything work fine.

My suggestion is to check the value in "Arduino Executable Path" and if it is directory, try to find arduino executable file in there. Another suggestion (more complex I know) is to try to looking for the path in default install path (in Linux for example usually is in $HOME/arduino-X-Y-Z .

Sorunome, 2018-01-04:
> My suggestion is to check the value in "Arduino Executable Path" and if it is directory, try to find arduino executable file in there. Another suggestion (more complex I know) is to try to looking for the path in default install path (in Linux for example usually is in $HOME/arduino-X-Y-Z .

If you are installing via a package manager, like you should be, then the executable will be in your $PATH and the default, `arduino` will work