Sorunome/arduino-upload #5: Uncaught TypeError: Cannot read property 'list' of null

pouya-barzegar, 2016-07-30:
[Enter steps to reproduce below:]

1. Go to the Tab with the file i want to upload or to the project
2. From packages section, click on arduino Upload -> Upload

**Atom Version**: 1.8.0
**System**: "Arch Linux"
**Thrown From**: [arduino-upload](https://github.com/Sorunome/arduino-upload) package, v0.7.1

### Stack Trace

Uncaught TypeError: Cannot read property 'list' of null

```
At /home/parch/.atom/packages/arduino-upload/lib/arduino-upload.coffee:190

TypeError: Cannot read property 'list' of null
    at Object.module.exports.ArduinoUpload.getPort (/home/parch/.atom/packages/arduino-upload/lib/arduino-upload.coffee:190:13)
    at Object.module.exports.ArduinoUpload.upload (/home/parch/.atom/packages/arduino-upload/lib/arduino-upload.coffee:148:5)
    at atom-workspace.subscriptions.add.atom.commands.add.arduino-upload:upload (/home/parch/.atom/packages/arduino-upload/lib/arduino-upload.coffee:68:33)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:260:29)
    at CommandRegistry.handleCommandEvent (/usr/share/atom/resources/app.asar/src/command-registry.js:3:61)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/usr/share/atom/resources/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/usr/share/atom/resources/app.asar/src/atom-environment.js:1100:28)
    at EventEmitter.outerCallback (/usr/share/atom/resources/app.asar/src/application-delegate.js:327:25)
    at emitTwo (events.js:87:13)
    at EventEmitter.emit (events.js:172:7)
```

### Commands

```
     -0:01.5.0 arduino-upload:upload (atom-text-editor.editor.vim-mode.power-mode.insert-mode.is-focused)
```

### Config

```json
{
  "core": {
    "audioBeep": false,
    "autoHideMenuBar": true,
    "destroyEmptyPanes": false,
    "disabledPackages": [
      "you-complete-me",
      "language-javascript",
      "autocomplete-java",
      "file-explorer",
      "build",
      "activate-power-mode",
      "autocomplete-bash-builtins",
      "autocomplete-clang",
      "platformio-ide",
      "platformio-ide-terminal"
    ],
    "excludeVcsIgnoredPaths": false,
    "packagesWithKeymapsDisabled": [],
    "projectHome": "/home/parch/Documents/projects/",
    "themes": [
      "atom-dark-ui",
      "monokai-solarized-dark"
    ]
  },
  "arduino-upload": {
    "board": "arduino:avr"
  }
}
```

### Installed Packages

```coffee
# User
activate-power-mode-delete, v0.6.1
arduino-upload, v0.7.1
autocomplete-haskell, v0.6.5
busy, v0.6.0
fun-files, v0.10.0
gpp-compiler, v3.0.4
haskell-ghc-mod, v1.16.0
ide-haskell, v1.6.5
ide-haskell-cabal, v1.4.1
ide-haskell-repl, v0.0.23
language-arduino, v0.4.1
language-assembly, v0.6.2
language-haskell, v1.7.18
language-ini, v1.16.0
linter, v1.11.14
linter-gcc, v0.6.15
monokai-solarized-dark, v0.1.1
pigments, v0.31.2
tool-bar, v1.0.0
vim-mode, v0.65.1

# Dev
No dev packages
```

Sorunome, 2016-07-30:
Thanks for reporting, this is because `serialport` fails to install. You are still able to build / verify normally. If you want to follow the resolving, check it out here: https://github.com/atom/apm/issues/599

Sorunome, 2016-09-07:
Fixed with https://github.com/Sorunome/arduino-upload/releases/tag/v1.0.0

Requires atom 1.10 or later

closed 2016-09-07 by Sorunome