Sorunome/arduino-upload #7: Error during installation

viktor1190, 2016-12-16:
I'm trying to install this package from Atom, but there is an problem and atom is showing a log error.
I try to install it by my self using the git-bash command line on Windows, then the same error appear again:

> Installing arduino-upload to C:\Users\DESARROLLO 5\.atom\packages failed

> serialport@4.0.7 install C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\node_modules\arduino-upload\node_modules\serialport
> node-pre-gyp install --fallback-to-build

> C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\node_modules\arduino-upload\node_modules\serialport>if not defined npm_config_node_gyp (node "C:\Users\DESARROLLO 5\AppData\Local\atom\app-1.12.7\resources\app\apm\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" clean )  else (node "" clean )
Failed to execute 'node-gyp.cmd clean' (1)
C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j
`-- (empty)
 node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-win32-ia32.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp)
module.js:327
    throw err;
    ^
Error: Cannot find module 'C:\Users\DESARROLLO 5\AppData\Local\atom\app-1.12.7\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp.cmd clean' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\node_modules\arduino-upload\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.10586
node-pre-gyp ERR! command "C:\\Users\\DESARROLLO 5\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\DESARR~1\\AppData\\Local\\Temp\\apm-install-dir-1161116-11232-88yf1j\\node_modules\\arduino-upload\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\node_modules\arduino-upload\node_modules\serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\package.json'
npm WARN apm-install-dir-1161116-11232-88yf1j No description
npm WARN apm-install-dir-1161116-11232-88yf1j No repository field.
npm WARN apm-install-dir-1161116-11232-88yf1j No README data
npm WARN apm-install-dir-1161116-11232-88yf1j No license field.
npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\Users\\DESARROLLO 5\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\DESARROLLO 5\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\DESARROLLO 5\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\DESARROLLO 5\\.atom\\.apmrc" "install" "C:\\Users\\DESARR~1\\AppData\\Local\\Temp\\d-1161116-11232-i84rpt\\package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=ia32" "--global-style"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.
npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\DESARR~1\AppData\Local\Temp\apm-install-dir-1161116-11232-88yf1j\npm-debug.log
npm ERR! code 1

I have node version 4.6.0 running on intel i7 windows 10 machine. I try to get the link that is showing in the above log: https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-win32-ia32.tar.gz but get a 404 not found page. Maybe that is the problem, I supose.

Sorunome, 2016-12-17:
I just tried, it still installs fine for me. What do you get when running `apm --version` ?

viktor1190, 2016-12-23:
$ apm --version
apm  1.12.9
npm  3.10.5
node 4.4.5
python 3.5.1
git 2.9.0.windows.1
visual studio

I have tried on Debian/Linux, Works fine.

Sorunome, 2017-02-02:
Is this issue still present or did some update randomly fix it?

Sorunome, 2017-03-26:
Duplicate of #8

closed 2017-03-26 by Sorunome