Sorunome/arduino-upload #8: Can't install

JamesCoyle, 2016-12-30:
```
> serialport@4.0.7 install C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport
> node-pre-gyp install --fallback-to-build

Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Platforms\Win32\PlatformToolsets\v140\Toolset.targets(34,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\build\serialport.vcxproj]
Failed to execute 'C:\Users\james\AppData\Local\atom\app-1.12.7\resources\app\apm\bin\node.exe C:\Users\james\AppData\Local\atom\app-1.12.7\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\build\Release' (1)
C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66
`-- (empty)

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-win32-ia32.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp) 
gyp ERR! build error 
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\james\AppData\Local\atom\app-1.12.7\resources\app\apm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Users\\james\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\james\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\Users\\james\\AppData\\Local\\Temp\\apm-install-dir-1161130-4256-y02f66\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\james\\AppData\\Local\\Temp\\apm-install-dir-1161130-4256-y02f66\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release"
gyp ERR! cwd C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\james\AppData\Local\atom\app-1.12.7\resources\app\apm\bin\node.exe C:\Users\james\AppData\Local\atom\app-1.12.7\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\build\Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 10.0.14393
node-pre-gyp ERR! command "C:\\Users\\james\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\james\\AppData\\Local\\Temp\\apm-install-dir-1161130-4256-y02f66\\node_modules\\arduino-upload\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\node_modules\arduino-upload\node_modules\serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\package.json'
npm WARN apm-install-dir-1161130-4256-y02f66 No description
npm WARN apm-install-dir-1161130-4256-y02f66 No repository field.
npm WARN apm-install-dir-1161130-4256-y02f66 No README data
npm WARN apm-install-dir-1161130-4256-y02f66 No license field.
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Users\\james\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\james\\AppData\\Local\\atom\\app-1.12.7\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\james\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\james\\.atom\\.apmrc" "install" "C:\\Users\\james\\AppData\\Local\\Temp\\d-1161130-4256-1imwxr5\\package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=ia32" "--global-style" "--msvs_version=2015"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\james\AppData\Local\Temp\apm-install-dir-1161130-4256-y02f66\npm-debug.log
npm ERR! code 1
```

```
apm --version
apm  1.12.9
npm  3.10.5
node 4.4.5
python 2.7.12
git 2.11.0.windows.1
visual studio 2015
```

JamesCoyle, 2017-01-07:
Just wondering if it has anything to do with the double backslash on this line after the`\apm\bin\\..\`:
```
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\james\AppData\Local\atom\app-1.12.9\resources\app\apm\bin\node.exe C:\Users\james\AppData\Local\atom\app-1.12.9\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build
```

Sorunome, 2017-01-07:
That shouldn't be an issue at all, the `\\`.
It seems to me to be the same thing as #7

JamesCoyle, 2017-01-07:
Just seemed possible that line was copied and pasted and then all the escape characters were removed except that one. I'm assuming it just gets ignored?

Sorunome, 2017-01-07:
Yeah, multiple slashes/backslashes following each other are ignored in file paths

Sorunome, 2017-02-02:
Is the issue still present or did some update randomly fix it?

JamesCoyle, 2017-02-02:
Same issue with version 1.0.3. Is there anything I need to have installed for it to work? I think I have pretty much everything I can think of installed already.

Sorunome, 2017-02-02:
I'm not sure, i'm pretty much blaiming windows ATM as linux users are being able to use it just fine....

Bardocz-Csaba, 2017-03-08:
Same error for me. :(

yzmotoxer807, 2017-03-25:
Any solution? I can not install and have the same errors.

Sorunome, 2017-03-26:
Python? This doesn't use any python at all....

Also, I don't have a windows machine :/

yzmotoxer807, 2017-03-26:
Yeah that serialport released binary path no longer exists. But after I installed that package manually from its own repo, it still didn't work until I installed python 2.7. Previously I had 3.6 installed and there is some dependency that relies on pre-3.0 python.

reconbot, 2017-03-26:
Node gyp required python

tejas007tez, 2017-03-31:
```> serialport@4.0.7 install C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport
> node-pre-gyp install --fallback-to-build

Failed to execute 'C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\bin\node.exe C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport\build\Release --msvs_version=2010' (1)
C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf
`-- (empty)

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-win32-ia32.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "C:\Users\Tejas\AppData\Local\Programs\Python\Python36\", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\node_modules\node-gyp\lib\configure.js:449:14)
gyp ERR! stack     at C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\node_modules\node-gyp\lib\configure.js:404:11
gyp ERR! stack     at C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:82:15)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Users\\Tejas\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Tejas\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Users\\Tejas\\AppData\\Local\\Temp\\apm-install-dir-117231-6076-1evpvzf\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\Tejas\\AppData\\Local\\Temp\\apm-install-dir-117231-6076-1evpvzf\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release" "--msvs_version=2010"
gyp ERR! cwd C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\bin\node.exe C:\Users\Tejas\AppData\Local\atom\app-1.15.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport\build\Release --msvs_version=2010' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Windows_NT 6.1.7601
node-pre-gyp ERR! command "C:\\Users\\Tejas\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Tejas\\AppData\\Local\\Temp\\apm-install-dir-117231-6076-1evpvzf\\node_modules\\arduino-upload\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\node_modules\arduino-upload\node_modules\serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\package.json'
npm WARN apm-install-dir-117231-6076-1evpvzf No description
npm WARN apm-install-dir-117231-6076-1evpvzf No repository field.
npm WARN apm-install-dir-117231-6076-1evpvzf No README data
npm WARN apm-install-dir-117231-6076-1evpvzf No license field.
npm ERR! Windows_NT 6.1.7601
npm ERR! argv "C:\\Users\\Tejas\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Tejas\\AppData\\Local\\atom\\app-1.15.0\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\Tejas\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\Tejas\\.atom\\.apmrc" "install" "C:\\Users\\Tejas\\AppData\\Local\\Temp\\d-117231-6076-19adhmd\\package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=ia32" "--global-style" "--msvs_version=2010"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
[... cut ...]
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Tejas\AppData\Local\Temp\apm-install-dir-117231-6076-1evpvzf\npm-debug.log
npm ERR! code 1```

NEED HELP

audstanley, 2017-04-05:
I Installed latest python 2.7.x after the same error (on windows 10) and it installed just fine.

Sorunome, 2017-09-01:
I did a lot of digging today, it should install now on windows!

closed 2017-09-01 by Sorunome

yokai117, 2017-09-13:
@Sorunome How can I tell it when it builds where my python 2.7 install is. 3.5 is my default and I would rather it stay that way.  Or how can I build these packages manually so that I can tell them where my python build is?  I'm on Windows 10.

yokai117, 2017-09-13:
Added python 2.7 before my python 3.5 in my path variable and apm was able to successfully download it.  I've managed to get something to build now.  Thanks!

Sorunome, 2017-09-13:
Glad you got it to work!
would you still mind to share your atom version (and the URL off of which it failed to download)? so that i can serve pre-builds for others using your configuration. Thanks!

yokai117, 2017-09-13:
My versions are: 
```.\bin\apm --version 
apm  1.18.4           
npm  3.10.10          
node 6.9.5 x64        
python 3.5.2          
git 2.9.3.windows.1   
visual studio 2015```

And my error is this:
```
 .\bin\apm install arduino-upload
Installing arduino-upload to C:\Users\alharris\.atom\packages failed

> serialport-builds-electron@4.0.7 install C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron
> node-pre-gyp install --fallback-to-build

Failed to execute 'C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\bin\node.exe C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release --msvs_version=2015' (1)
C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9
`-- (empty)

node-pre-gyp ERR! Tried to download(undefined): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport-builds-electron@4.0.7 and electron@1.6.9 (electron-v1.6 ABI) (falling back to source compile with node-gyp)
gyp ERR! configure error
gyp ERR! stack Error: Python executable "C:\Users\alharris\AppData\Local\Programs\Python\Python35-32\python.EXE" is v3.5.2, which is not supported by gyp.
gyp ERR! stack You can pass the --python switch to point to Python >= v2.5.0 & < 3.0.0.
gyp ERR! stack     at failPythonVersion (C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\node_modules\node-gyp\lib\configure.js:454:14)
gyp ERR! stack     at C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\node_modules\node-gyp\lib\configure.js:443:9
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:197:7)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
gyp ERR! stack     at Socket.<anonymous> (internal/child_process.js:334:11)
gyp ERR! stack     at emitOne (events.js:96:13)
gyp ERR! stack     at Socket.emit (events.js:188:7)
gyp ERR! stack     at Pipe._handle.close [as _onclose] (net.js:498:12)
gyp ERR! System Windows_NT 10.0.10586
gyp ERR! command "C:\\Users\\alharris\\AppData\\Local\\atom\\app-1.20.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\alharris\\AppData\\Local\\atom\\app-1.20.0\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Users\\alharris\\AppData\\Local\\Temp\\1\\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\\node_modules\\arduino-upload\\node_modules\\serialport-builds-electron\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\alharris\\AppData\\Local\\Temp\\1\\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\\node_modules\\arduino-upload\\node_modules\\serialport-builds-electron\\build\\Release" "--msvs_version=2015"
gyp ERR! cwd C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\bin\node.exe C:\Users\alharris\AppData\Local\atom\app-1.20.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release --msvs_version=2015' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.10586
node-pre-gyp ERR! command "C:\\Users\\alharris\\AppData\\Local\\atom\\app-1.20.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\alharris\\AppData\\Local\\Temp\\1\\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\\node_modules\\arduino-upload\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\node_modules\arduino-upload\node_modules\serialport-builds-electron
node-pre-gyp ERR! node -v v6.9.5
[... cut ...]
npm ERR!     npm bugs serialport-builds-electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport-builds-electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\alharris\AppData\Local\Temp\1\apm-install-dir-117813-2908-crn6hc.5rfuf5stt9\npm-debug.log
npm ERR! code 1
```
My Installation continually failed on the python requirement but it does appear to be using your new location.

Sorunome, 2017-09-13:
huh, for me the download is working..... what happens when you visit https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-win32-x64.tar.gz manually?

yokai117, 2017-09-13:
So the bizarre part is that I can hit the server http://144.76.31.180/ but I can't seem to hit the HTTPS site.  When I switched to http it worked but my firewall blocked it.  This seems to be on my end that's blocking it.

Sorunome, 2017-09-13:
maybe you lack the letsencrypt root cert?

yokai117, 2017-09-13:
It's actually a firewall issue.  I'll look into it later and add an explicit trust.  I would assume that LetsEncrypt site uses their own certificates.  Also I'm not getting a certificate error I'm getting a connection error.

Sorunome, 2017-09-13:
connection error could also mean you use ipv6 and for some reason files.sorunome.de isn't working with that....so, are you using ipv6?

yokai117, 2017-09-13:
There's a firewall on my system that causes all sorts of these problems.  I'll make changes later to it and see if that corrects it.

Sorunome, 2017-09-13:
Ah ok ^.^