Sorunome/arduino-upload #10: Package can't find Arduino IDE on Linux

mehalom, 2017-01-29:
Program crashes with `'arduino' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process.`

Sorunome, 2017-01-29:
You need to install the arduino IDE. You won't actually be using the IDE, but it needs to be installed.

Sorunome, 2017-01-29:
What does `which arduino` output you?

mehalom, 2017-01-29:
Sorry, I did't catch. What is it?

> `which arduino` output

Sorunome, 2017-01-29:
Try running
```
which arduino
```
in the terminal and post the ouput ^.^

mehalom, 2017-01-29:
nothing
 It seems, that the program can't find manually installed Arduino IDE

Sorunome, 2017-01-29:
Ok, in the package settings edit "Arduino Executable Path" to the absolute path to where the executable of your manual-installed IDE lies.

BTW, why won't you just install the arduino IDE via the repo? That would make things easier ;)

mehalom, 2017-01-29:
I've installed IDE manually because Ubuntu's repos are very outdated.(

Sorunome, 2017-01-29:
That's because of the release cycles......anyhow, did you get it to work now?

mehalom, 2017-01-30:
Not it crashes with `Uncaught Error: spawn EACCES`. (

Sorunome, 2017-01-30:
do you have any more information on that error? Are you sure you entered the path to the *executable*, not just the folder where it is in?

mehalom, 2017-01-30:
Rechecked folder, entered path directly to .desktop file. Now it just doesn't recognize connected arduino board (Arduino IDE itself connects normally).

Sorunome, 2017-01-30:
noooo, you don't enter the path to the .desktop file, you enter the path to the actual executable!

For finding that out you can open your .desktop file with a text editor, somewhere you will have something like `Exec=<something>`, that `<something>` is exactly the path that you'll need to put into the settings

mehalom, 2017-01-30:
Same issue. Compiler works, but Uploader and Serial monitor not. (Sorry for my stupitidy)

Sorunome, 2017-01-30:
This is not your stupidity, what you just mentioned might be an actual bug.

May I have the output of
```
lsusb
```
please, while the arduino is plugged in? You are free to only share the arduino line thing.

mehalom, 2017-01-30:
`Bus 002 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 005 Device 002: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Bus 003 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub`
Arduino definitely plugged in and IDE uploads new sketches as usual.

Sorunome, 2017-01-30:
uuuuh, i can't see from the names which one the arduino is, so could you please send me the output of `lsusb` while the arduino is **not** plugged in additonally?

mehalom, 2017-01-30:
`Bus 003 Device 003: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter`
This is the only difference between plugged and unplugged Arduino.

Sorunome, 2017-01-30:
It should be fixed now, there should be a new update. If it is still broken please create a new issue with the full error message and stuff ^.^

commit 7606b95d (2017-01-30): fixes #10
closed 2017-01-30 by Sorunome