Sorunome/arduino-upload #11: can't install

Hartze, 2017-01-31:
I’m failing at install this Package via the Package installer. 
This is my Output:
`
make: Verzeichnis »/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build« wird betreten
  CXX(target) Release/obj.target/serialport/src/serialport.o
make: Verzeichnis »/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build« wird verlassen
Failed to execute '/usr/share/atom/resources/app/apm/bin/node /usr/share/atom/resources/app/apm/bin/../node_modules/.bin/node-gyp build --fallback-to-build --module=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
/tmp/apm-install-dir-117031-5601-u55wao
└── (empty)

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp) 
make: g++: Befehl nicht gefunden
make: *** [Release/obj.target/serialport/src/serialport.o] Fehler 127
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/atom/resources/app/apm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.19.0-32-generic
gyp ERR! command "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/.bin/node-gyp" "build" "--fallback-to-build" "--module=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release"
gyp ERR! cwd /tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/share/atom/resources/app/apm/bin/node /usr/share/atom/resources/app/apm/bin/../node_modules/.bin/node-gyp build --fallback-to-build --module=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 3.19.0-32-generic
node-pre-gyp ERR! command "/usr/share/atom/resources/app/apm/bin/node" "/tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /tmp/apm-install-dir-117031-5601-u55wao/node_modules/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/tmp/apm-install-dir-117031-5601-u55wao/package.json'
npm WARN apm-install-dir-117031-5601-u55wao No description
npm WARN apm-install-dir-117031-5601-u55wao No repository field.
npm WARN apm-install-dir-117031-5601-u55wao No README data
npm WARN apm-install-dir-117031-5601-u55wao No license field.
npm ERR! Linux 3.19.0-32-generic
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/johannes/.atom/.apm/.apmrc" "--userconfig" "/home/johannes/.atom/.apmrc" "install" "/tmp/d-117031-5601-yg13ny/package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=x64" "--global-style"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-117031-5601-u55wao/npm-debug.log
npm ERR! code 1
`
This is Output of apm --version:
`
apm  1.12.9
npm  3.10.5
node 4.4.5
python 2.7.6
git 1.9.1
`

Sorunome, 2017-02-01:
Your apm, npm and node version seem quite out-of-date, try updating maybe?

Hartze, 2017-02-02:
I now compiled from source using [this](https://github.com/atom/atom/blob/master/docs/build-instructions/linux.md) guide. 
the Output of "apm --version" is now:
```
apm  1.15.3
npm  3.10.5
node 4.4.5 x64
python 2.7.6
git 1.9.1
```
the installation worked like a charm, thx.

I am new to git-hub if I have a other problem can I mention it here, or have I to open a new issue, or there other ways to ask questions? If I’m wrong her pleas correct me.

However if I am create a new file in Atom (like test.ino) I get the "File isn't part of an Arduino sketch!" error. If I’m create a File via the arduinoIDE verify/build/upload works fine. are there other ways to create a sketch than my "workaround"?

Sorunome, 2017-02-02:
This is because of how arduino sketches look like (The arduino IDE does it automatically for you):

If you create a file `test.ino` and you want it to be a new sketch, that file must be located in a folder called `test`. All other .ino files in that folder will be auto-included. So you'll actually have to create the file `test/test.ino`, that should work, then.

Hartze, 2017-02-02:
Works! thx

closed 2017-02-02 by Hartze