Sorunome/arduino-upload #12: Qutestion about Error console when Verify/Build/Upload

Hartze, 2017-02-03:
Hi me again,
when I'm trying to Verify/Build/Upload I only the notification in the upper right coner, not the console on the bottom witch contains the error messages.
Both if I'm start it via drop down menu or the command pallete.
Does I have to activate it anywhere?

Sorunome, 2017-02-03:
As long as there is no error during verify/build/upload the error console shouldn't open, as soon as there is an error the console will open itself automatically.

You can try this by placing an obvious syntax error somewhere.

Hartze, 2017-02-03:
my code is (sad enough) full of errors :/ 
I only get the Notification Build failed.

Sorunome, 2017-02-03:
Is the code too long to post? 'cause I'm smelling a bug in this atom plugin

Hartze, 2017-02-03:
Not to long but to bad for public.
I also don’t get a console with the default examples.
Here I remove the Semicolon in Line 27:
```
/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino model, check
  the Technical Specs of your board  at https://www.arduino.cc/en/Main/Products

  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald

  modified 2 Sep 2016
  by Arturo Guadalupi

  modified 8 Sep 2016
  by Colby Newman
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT)
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

```

Sorunome, 2017-02-03:
looking at the code atm i can see why uploading may not display the error console, but that is definitally also the case with verify? Because for me it shows up on that one....which Arduino IDE are you using?

If that also happens with verifying, please compile in the IDE with verbose mode during compilation on and paste the output here ^.^

Hartze, 2017-02-03:
I'm not relay have tried to upload with this package, because for now i have it to copy it in the ArduinoIDE for debugging anyway, so for me only when I'm Verify/Build. 
I just tried it now, same on Upload.
My IDE-Version is: 1.8.1 (Downloaded from arduino.cc yesterday)
Output when Verifying with IDE Preferences:
Show verbose output during:
    - compilation: TRUE
    - upload: FALSE
Compiler warnings: NONE

```
Arduino: 1.8.1 (Linux), Board: "Arduino/Genuino Uno"

/home/johannes/Downloads/arduino-1.8.1/arduino-builder -dump-prefs -logger=machine -hardware /home/johannes/Downloads/arduino-1.8.1/hardware -tools /home/johannes/Downloads/arduino-1.8.1/tools-builder -tools /home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -built-in-libraries /home/johannes/Downloads/arduino-1.8.1/libraries -libraries /home/johannes/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path /tmp/arduino_build_236865 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -verbose /home/johannes/Arduino/exaplefalsch/exaplefalsch.ino
/home/johannes/Downloads/arduino-1.8.1/arduino-builder -compile -logger=machine -hardware /home/johannes/Downloads/arduino-1.8.1/hardware -tools /home/johannes/Downloads/arduino-1.8.1/tools-builder -tools /home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -built-in-libraries /home/johannes/Downloads/arduino-1.8.1/libraries -libraries /home/johannes/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path /tmp/arduino_build_236865 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr -verbose /home/johannes/Arduino/exaplefalsch/exaplefalsch.ino
Using board 'uno' from platform in folder: /home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr
Detecting libraries used...
"/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_236865/sketch/exaplefalsch.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_236865/sketch/exaplefalsch.ino.cpp" -o "/tmp/arduino_build_236865/preproc/ctags_target_for_gcc_minus_e.cpp"
"/home/johannes/Downloads/arduino-1.8.1/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/tmp/arduino_build_236865/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/home/johannes/Downloads/arduino-1.8.1/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/cores/arduino" "-I/home/johannes/Downloads/arduino-1.8.1/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_236865/sketch/exaplefalsch.ino.cpp" -o "/tmp/arduino_build_236865/sketch/exaplefalsch.ino.cpp.o"
/home/johannes/Arduino/exaplefalsch/exaplefalsch.ino: In function 'void setup()':
exaplefalsch:28: error: expected ';' before '}' token
 }
 ^
exit status 1
expected ';' before '}' token
```

Sorunome, 2017-02-03:
Ok, ummmm....would you mind giving me the output of something like
```
arduino /tmp/test/test.ino --verify
```

Where `arduino` is the full path to your arduino executable and `/tmp/test/test.ino` is the full path to the ino file

Hartze, 2017-02-03:
Output is:
```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/home/johannes/Arduino/exaplefalsch/exaplefalsch.ino: In function 'void setup()':
exaplefalsch:28: error: expected ';' before '}' token
 }
 ^
exit status 1

```

Sorunome, 2017-02-03:
OK, this is very weird.......

um, mind changing the source of the plugin in the following way:

The file `~/.atom/packages/arduino-upload/lib/arduino-upload.coffee`

There around line 114 you should see

```
			stdoutput.stderr.on 'data', (data) ->
				if data.toString().strip() == "exit status 1"
					dispError = false
				if dispError
					output.addLine data.toString(), workpath
				if data.toString().strip() == "Verifying..."
					dispError = true

```

Please insert a `console.log´ statement to make it look like this:
```
			stdoutput.stderr.on 'data', (data) ->
				console.log data.toString()
				if data.toString().strip() == "exit status 1"
					dispError = false
				if dispError
					output.addLine data.toString(), workpath
				if data.toString().strip() == "Verifying..."
					dispError = true
```

Now restart atom, hit ctrl+shift+i to open the debug console and verify again, giving me the output.

Hartze, 2017-02-03:
OK I also Installed [this](https://atom.io/packages/language-arduino) package and may I should have tried this earlier but when I'm disable it I get the console.
Give me 5 minutes and I'm change the file as you mentioned.

Sorunome, 2017-02-03:
That is weird, I use that package, too, and stuff is working for me o.o

Hartze, 2017-02-03:
I re-enable it and get the the console now.
edited the file now, output is:
```
arduino-upload.coffee [sm]:115 Picked up JAVA_TOOL_OPTIONS: 

arduino-upload.coffee [sm]:115 Loading configuration...

arduino-upload.coffee [sm]:115 Initializing packages...

arduino-upload.coffee [sm]:115 Preparing boards...

arduino-upload.coffee [sm]:115 Verifying...

arduino-upload.coffee [sm]:115 /home/johannes/workspace/example/example.ino: In function 'void setup()':

arduino-upload.coffee [sm]:115 example:28: error: expected ';' before '}' token
 }
 ^

arduino-upload.coffee [sm]:115 exit status 1
```
still get the console.

Sorunome, 2017-02-03:
dangit, this is very weird......mind reverting that change, and in `output-view.coffee` in the same folder change this: (around line 42)
```
		finish: ->
			if @message.trim() == ''
				@message = ''
				@hide()
				return
```
to this
```
		finish: ->
			console.log @message
			if @message.trim() == ''
				@message = ''
				@hide()
				return
```
And restart atom and post the output again?

Hartze, 2017-02-03:
I don't get a console when Uploading, as you assumed earlier. Output in debugging console when upload a Sketch with syntax error is:
`Object {comName: "/dev/ttyACM1", manufacturer: "Arduino_Srl", serialNumber: "Arduino_Srl_Arduino_Uno_9553034323535180C152", pnpId: "usb-Arduino_Srl_Arduino_Uno_9553034323535180C152-if00", vendorId: "0x2a03"…}`

after the change when Verifying:

```
<a data-line="-1" data-file="/home/johannes/workspace/example/example.ino">/home/johannes/workspace/example/example.ino:</a> In function 'void setup()':
<a data-line="28" data-file="/home/johannes/workspace/example/example.ino">example:28</a>: error: expected ';' before '}' token
 }
 ^

```
still get the console

Sorunome, 2017-02-03:
oh, wait, you _do_ get the console now?

I thought you were running these tests on when you didn't get the console?

Hartze, 2017-02-03:
No after disable and re-enable of the language-arduino package I get the console. As I mentioned before 20 Minutes. I thought I'm helping you finding a Bug.

Sorunome, 2017-02-03:
I'm so sorry i did't catch that line and i htought the console still didn't appear for you D:

If you want to help some more you could try to get it again where the console doesn't appear, or, if not, just close this issue.

Again, I am so sorry, i thought the console still didn't open for you D:

Hartze, 2017-02-03:
I already try to reproduce the bug, but I'm not able so far. I've done 2 things before the console appears:

1. run this Shell command: 
`Downloads/arduino-1.8.1/arduino Arduino/exaplefalsch/exaplefalsch.ino --verify`
2. Disabled the language-arduino package

Absolutely no problem.
You are the one who spending his/her/apache time to get me the console.
The Arduino Community is in my opportunity the Idea of working freely  together over the Internet.
If you have any Idea how I can reproduce this Bug I'm happy to try it.

Sorunome, 2017-02-03:
So the console still shows up with the `language-arduino` package enabled? If so, please just close this issue, perhaps somebody else will stumble upon it and i'll be able to fix it then, heehee

Other than that, thank you for taking your time trying to help ^.^

closed 2017-02-03 by Hartze