Sorunome/arduino-upload #13: default path on macOS

bgentry, 2017-02-18:
AFAIK, installing the Arduino IDE on macOS consists solely of putting the `.app` bundle into `/Applications`. At no point is an `arduino` binary added to the user's `$PATH`. Digging through the app bundle, it looks like all there is is an `Arduino` binary which launches the IDE.

Perhaps on macOS this library should instead default to expecting the executable path to be `/Applications/Arduino.app/Contents/MacOS/Arduino`? That at least gets the verify and upload commands to run, though serial monitor doesn't seem to work properly.

Sorunome, 2017-02-18:
Would only `Arduino` work too? So is that thing in the $PATH? I'd prefer from reverting to stuff in $PATH rather than absolute paths

bgentry, 2017-02-18:
Nope, it doesn't get added to the `$PATH`. If I install via homebrew, then I get `arduino-builder` on the `$PATH`, but that doesn't seem to be the right tool. Now that I tried it, uploading actually doesn't work with the path value I suggested anyway. Only verify seems to work. Not sure how to resolve it and couldn't find any instructions for getting a proper CLI binary on macOS.

Sorunome, 2017-02-18:
Which version does the macOS one use?

BTW please note that i only run linux and can test neither macOS nor windows <_<

bgentry, 2017-02-18:
> Which version does the macOS one use?

Which version of what? If you mean `arduino-builder`, it's:

```
➜   arduino-builder --version
Arduino Builder 1.3.23
Copyright (C) 2015 Arduino LLC and contributors
See https://www.arduino.cc/ and https://github.com/arduino/arduino-builder/graphs/contributors
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

I have Arduino version 1.8.1 installed. Again I think the `arduino-builder` install on the `$PATH` might be specific to those who've installed via Homebrew, which is likely a small minority of overall Arduino macOS users. I believe that binary exists for all users at `/Applications/Arduino.app/Contents/Java/arduino-builder`.

Sorunome, 2017-02-23:
Do you happen to know if this is the same as #4 ? So does it actually output that error?

bgentry, 2017-02-23:
No, there are no crashes or errors like that at all.

Sorunome, 2017-09-01:
I added default paths for macos and windows recently. https://github.com/Sorunome/arduino-upload/blob/f8f3586385c6de24b93b941214820bd02ef2ac82/lib/util.coffee line 5 to 13

closed 2017-09-01 by Sorunome