Sorunome/arduino-upload #15: Can not install the plugin

juwul, 2017-03-24:
```
Last login: Thu Mar 23 17:53:58 on ttys000
Juris-MacBook-Pro:~ juriwulff$ apm install arduino-upload
Installing arduino-upload to /Users/juriwulff/.atom/packages ✗

> serialport@4.0.7 install /private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport
> node-pre-gyp install --fallback-to-build

Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld
└── (empty)

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-darwin-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: Python executable "/Applications/Atom.app/Contents/Resources/app/apm/bin/python-interceptor.sh" is v3.6.0, which is not supported by gyp.
gyp ERR! stack You can pass the --python switch to point to Python >= v2.5.0 & < 3.0.0.
gyp ERR! stack     at failPythonVersion (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/lib/configure.js:454:14)
gyp ERR! stack     at /Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/lib/configure.js:443:9
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:204:7)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
gyp ERR! stack     at Socket.<anonymous> (internal/child_process.js:319:11)
gyp ERR! stack     at emitOne (events.js:77:13)
gyp ERR! stack     at Socket.emit (events.js:169:7)
gyp ERR! stack     at Pipe._onclose (net.js:477:12)
gyp ERR! System Darwin 16.4.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/.bin/node-gyp" "configure" "--fallback-to-build" "--module=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release"
gyp ERR! cwd /private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:827:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Darwin 16.4.0
node-pre-gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/node_modules/arduino-upload/node_modules/serialport
node-pre-gyp ERR! node -v v4.4.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/package.json'
npm WARN apm-install-dir-117224-27089-1ye0lld No description
npm WARN apm-install-dir-117224-27089-1ye0lld No repository field.
npm WARN apm-install-dir-117224-27089-1ye0lld No README data
npm WARN apm-install-dir-117224-27089-1ye0lld No license field.
npm ERR! Darwin 16.4.0
npm ERR! argv "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/juriwulff/.atom/.apm/.apmrc" "--userconfig" "/Users/juriwulff/.atom/.apmrc" "install" "/private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/d-117224-27089-zndc1p/package.tgz" "--runtime=electron" "--target=1.3.13" "--arch=x64" "--global-style"
npm ERR! node v4.4.5
npm ERR! npm  v3.10.5
npm ERR! code ELIFECYCLE

npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/7n/6slhynld2tdc65n5p_yzcvrc0000gn/T/apm-install-dir-117224-27089-1ye0lld/npm-debug.log
npm ERR! code 1

Juris-MacBook-Pro:~ juriwulff$ 

```

juwul, 2017-03-24:
Ahh just saw there this is reported in #8

Sorunome, 2017-03-25:
duplicate

closed 2017-03-25 by Sorunome