Sorunome/arduino-upload #19: Unable to install Arduino Upload

Ragdehl, 2017-04-16:
Hello,

I'm trying to install the Arduino Upload Package for Atom IDE but it does not work. When I press the Install button it does not start the dowload.

Thanks for your help.

Edgar

Sorunome, 2017-04-16:
Hi there, which OS and atom versions are you using? Is there any error output?

Ragdehl, 2017-04-16:
Hi I just installed it today, it is 1.16.0 versión. No error message appears. Only the "traffic forbidden" symbol appears after I press the "Install" button
[image]

Sorunome, 2017-04-16:
what happens if you run `apm install arduino-upload` in cmd?

Ragdehl, 2017-04-16:
Still an error appears:
[image]

Sorunome, 2017-04-16:
Try installing python 2.7.x https://www.python.org/downloads/

Ragdehl, 2017-04-16:
I installed Python 2.7 but still not working:

Installing “arduino-upload@1.0.3” failed.Hide output…

> serialport@4.0.7 install C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport
> node-pre-gyp install --fallback-to-build

Los proyectos de esta soluci�n se van a compilar de uno en uno. Para habilitar la compilaci�n en paralelo, agregue el modificador "/m".
MSBUILD : error MSB4132: No se reconoce la versi�n de herramientas "2.0". Las versiones de herramientas disponibles son "4.0".
Failed to execute 'C:\Users\Edgar\AppData\Local\atom\app-1.16.0\resources\app\apm\bin\node.exe C:\Users\Edgar\AppData\Local\atom\app-1.16.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport\build\Release' (1)
C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor
`-- (empty)

node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-electron-v1.3-win32-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and electron@1.3.13 (electron-v1.3 ABI) (falling back to source compile with node-gyp) 
gyp ERR! build error 
gyp ERR! stack Error: `C:\Windows\Microsoft.NET\Framework\v4.0.30319\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\Edgar\AppData\Local\atom\app-1.16.0\resources\app\apm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.15063
gyp ERR! command "C:\\Users\\Edgar\\AppData\\Local\\atom\\app-1.16.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Edgar\\AppData\\Local\\atom\\app-1.16.0\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "build" "--fallback-to-build" "--module=C:\\Users\\Edgar\\AppData\\Local\\Temp\\apm-install-dir-117316-15020-1a5enok.0akngeqaor\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\Edgar\\AppData\\Local\\Temp\\apm-install-dir-117316-15020-1a5enok.0akngeqaor\\node_modules\\arduino-upload\\node_modules\\serialport\\build\\Release"
gyp ERR! cwd C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\Edgar\AppData\Local\atom\app-1.16.0\resources\app\apm\bin\node.exe C:\Users\Edgar\AppData\Local\atom\app-1.16.0\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport\build\Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.15063
node-pre-gyp ERR! command "C:\\Users\\Edgar\\AppData\\Local\\atom\\app-1.16.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Edgar\\AppData\\Local\\Temp\\apm-install-dir-117316-15020-1a5enok.0akngeqaor\\node_modules\\arduino-upload\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\node_modules\arduino-upload\node_modules\serialport
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\package.json'
npm WARN apm-install-dir-117316-15020-1a5enok.0akngeqaor No description
npm WARN apm-install-dir-117316-15020-1a5enok.0akngeqaor No repository field.
npm WARN apm-install-dir-117316-15020-1a5enok.0akngeqaor No README data
npm WARN apm-install-dir-117316-15020-1a5enok.0akngeqaor No license field.
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Users\\Edgar\\AppData\\Local\\atom\\app-1.16.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Edgar\\AppData\\Local\\atom\\app-1.16.0\\resources\\app\\apm\\node_modules\\npm\\bin\\n [... cut ...]
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Edgar\AppData\Local\Temp\apm-install-dir-117316-15020-1a5enok.0akngeqaor\npm-debug.log
npm ERR! code 1

Sorunome, 2017-07-06:
Might be related with #8

Sorunome, 2017-09-01:
It should install now on windows!

closed 2017-09-01 by Sorunome