Sorunome/arduino-upload #20: can't upload with arduino-upload

sankerspace, 2017-04-26:
Hi,
The arduino-upload package is a cool stuff if someone want to work Atom-IDE,
Iam able to verify and build for my Arduino Pro Mini 3V , but if I want to upload
something goes wrong "Build error" (whereas Building alone with arduino-upload:build works).
Serial Monitor also works. Iam little confused, I also keep blank the board settings, to get settings out from the arduino environment(where uploads works).
Someone has an idea what is wrong here?

Sorunome, 2017-04-26:
I guess you mean this arduino? https://www.arduino.cc/en/Main/arduinoBoardProMini

Do you have multiple arduinos plugged in at the same time? That could lead to some issues.

And how do you usually program it anyways? Programming via things such as the ICE debugger isn't working yet....

Sorunome, 2017-09-01:
Is this issue still present?

Sorunome, 2018-03-19:
closing due to no replies. It may have been an arduino IDE bug that doesn't exist anymore....who knows?

Also i fixed since this issue quite a few things with uploading

closed 2018-03-19 by Sorunome