Sorunome/arduino-upload #21: Error output when compiling doesn't show

HackerInside0, 2017-05-08:
Atom 1.16.0 x64
Linux Mint 18.1 x64
Arduino 1.8.2

Sorunome, 2017-07-06:
I can't reproduce, do you have any more ifnormation?

qub1750ul, 2017-07-09:
In the meantime, i updated Arduino to 1.8.3 x64, Atom to 1.18.0 x64, and reinstalled the package in this moment.
Now it didn't work at all.
I've wrote the blink example in a new file created in Atom, then through the command palette i tried all the commands provided by the package.
In every case the result is the disappearance of the arduino-upload commands in the command palette until i disable and re-enable the package, there isn't any output message from the package or atom itself about this.
Using arduino by the CLI instead, works perfectly.

Sorunome, 2017-07-09:
Do you see any errors in the debugging console? (Ctrl+Shift+i)
What does the output when manually invocing via cli look like?

qub1750ul, 2017-07-09:
Just now Atom notified me about an incompatibility of the package becasue "build for a previous version of Atom", and offered me the possibility to fix it by clicking a button.
Doing it actually made the package to work again.
But when i introduce an error in the program the only output related to the error that i recive is "Build failed".

Opening the debug console i've found this error reported twice:

```
Could not connect to indexedDB  -> Event
/home/andrei/atom-1.11.1/out/app/node_modules/tree-view/lib/file.coffee:15378
EventdbOpenRequest.onerror @ /home/andrei/atom-1.11.1/out/app/node_modules/tree-view/lib/file.coffee:15378
```

When invoking arduino via cli the output is:
- with a good program:
```
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 4130 bytes (14%) of program storage space. Maximum is 28672 bytes.
Global variables use 149 bytes (5%) of dynamic memory, leaving 2411 bytes for local variabl
es. Maximum is 2560 bytes.
Uploading...
```
- with a program with an error:
```
Picked up JAVA_TOOL_OPTIONS:
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
/tmp/upino/upino.ino: In function 'void loop()':
upino:8: error: 'gg' was not declared in this scope
  gg;
  ^
exit status 1
```

Sorunome, 2017-07-09:
Do you use https://atom.io/packages/tree-view by any chance? If so, is the bug still present if you disable that?
Thank you for your help in debugging this!

qub1750ul, 2017-07-09:
Disabling the tree-view package compile errors are shown

Sorunome, 2017-07-09:
so you disable the tree-view package and then things work as expected? Weird, I am still unable to reproduce your issue......

With exactly which command did you invoke compiling via CLI?

qub1750ul, 2017-07-09:
arduino --upload --board arduino:avr:micro --port /dev/ttyACM0 upino.ino

qub1750ul, 2017-07-09:
In the debug console there are:

- This error reported twice, that is present since the startup of Atom and despite the tree-view package is disabled

```
Could not connect to indexedDB  -> Event
/home/andrei/atom-1.11.1/out/app/node_modules/tree-view/lib/file.coffee:15378
EventdbOpenRequest.onerror @ /home/andrei/atom-1.11.1/out/app/node_modules/tree-view/lib/file.coffee:15378
```

- And then this message that appears when i use any arduino-upload command by the command palette with a program that contains an error. It reflects the message that i get in a box at the bottom of the editor

```
arduino-upload.coffee [sm]:143 

/tmp/upino/upino.ino: In function 'void loop()':
upino:8: error: 'gg' was not declared in this scope
  gg;
  ^
```

- When i launch any arduino-upload command by the command palette with a good program, no other output is generated in the debugging console, and the only visible output are some notifications who inform me about the compilation/upload status.

Sorunome, 2017-07-09:
Okay so the arduino-upload package detects the error message correctly and buffers it internally....for some reason doesn't display it, though.

Which theme packages do you use? It might not be able to create the text output box....

qub1750ul, 2017-07-09:
I don't think that the theme is the problem.
I've done a test both with the theme that i use recently ( a customized version of Atom Material ) and the theme that i used when i opened the issue ( One Dark ), with the tree-view disabled both work fine:

Some screenshots:
[Material - error](https://www.dropbox.com/s/zdoev34ci6lqt9v/editor-error.png?dl=0)
[One Dark - error ](https://www.dropbox.com/s/xwpr9jqvj2vxqs3/one-dark-error.png?dl=0l)

[Material - succesful](https://www.dropbox.com/s/wwge0vpqq40lmlo/editor-ok.png?dl=0)
[One Dark - succesful](https://www.dropbox.com/s/v2s4sxv2a2u86ya/one-dark-fine.png?dl=0)

qub1750ul, 2017-07-09:
...and after i enable the tree-view package package to do a test it magically works... WTF?

Sorunome, 2017-07-09:
Soooo.....does it reliably work when restarting atom now, magically?

qub1750ul, 2017-07-09:
Yes, omitting that when the sketch is open in preview mode the message "File isn't part of an Arduino sketch!" appears, all works fine.

Sorunome, 2017-07-09:
Okay cool, I'll close this issue then! If you have the feeling that the preview mode one needs a new issue ticket thing then please go ahead and open a new bug report ^.^

My guess is that atom somehow screwed up a bit with version upgrading and disabling- and enabling that tree-view package fixed some stuff.

closed 2017-07-09 by Sorunome