Sorunome/arduino-upload #24: no arduino connected

jgramsch, 2017-08-15:
Hi! 
I'm getting the messaage "no arduino connected" even tho I can build the sketch and my arduino ide is set to the port. 
I'm using ubuntu gnome, arduino ide 1.8.4, and arduino-upload 1.1.0, atom 1.19.
I haven't been able to find where to set the port in the package. 
Help pls!

Sorunome, 2017-08-15:
Hello there, unlike the IDE arduino-upload automatically detects which port to use. Unfortunatelly this doesn't work yet for e.g. the arduino zero direct usb port.
That being said, which arduino are you using?

jgramsch, 2017-08-15:
I'm using the arduino yún, the port is connected to is /dev/ttyACM0 i think

Sorunome, 2017-08-15:
The port name sounds like it is the same issue, i'll look at it in a bit

Sorunome, 2017-08-15:
Okay, this might be something else as it turns out arduino zero is actually working (i must have forgotten / missed when i fixed that ^^)

Anyhow, would you mind giving me the output of `lsusb` once while the arduino is plugged in and once while it isn't? It might be that the vendor / product ID isn't registered yet

jgramsch, 2017-08-15:
**Bus 002 Device 012: ID 2341:8041 Arduino SA** 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 04f2:b328 Chicony Electronics Co., Ltd 
Bus 001 Device 005: ID 0489:e036 Foxconn / Hon Hai 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 004: ID 0458:0056 KYE Systems Corp. (Mouse Systems) Ergo 300 Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

jgramsch, 2017-08-15:
I highlighted the arduino (i think) thanks for all your help so far!

Sorunome, 2017-08-15:
Okay, that board should be known to this extension. Mind posting the output when uploading via the arduino IDE? You have to enable verbose output in the settings (`ctrl+,`)

disperso, 2017-08-26:
I have the same issue. Mine is listed as this in lsusb:

Bus 001 Device 003: ID 2341:0043 Arduino SA Uno R3 (CDC ACM)

In the code I saw that vendor 2341 is accepted for all product ids, so as you said it must be something else.

Thank you very much for this package. Although I haven't been able to use it yet, I

Sorunome, 2017-08-26:
This sounds like both of you are using samd based boards.
Are you uploading using the programmer port or the direct USB port? What is the verbose output when uploading via the arduino IDE?

disperso, 2017-08-26:
This is what I think is the relevant part of the verbose output:

`

         Using Port                    : /dev/ttyACM0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

`

Sorunome, 2017-08-29:
Soooooo, I just pushed a bigfix where, if the code has a syntax error, it claimed there was no arduino connected. Maybe your code actually only had a syntax error? Does it work now? I also provided a (for me) more meaningful debug output in the console, opened via ctrl+shift+i. If the bug still persists, please paste me an output from there!

PissedWizard, 2017-12-05:
Hi,

Hey got my atom worked out and have the latest version of it and arduino_upload installed on win7.
Atom works fine and can verify and build no problem.
I can not upload or use the serial monitor even though my arduino mega2560 shows up as com5 in the devMangr. It just does not see it.
Can you help?

PissedWizard, 2017-12-08:
Never mind, backed out a couple programs with a restore and it started working.

StaticRocket, 2018-11-02:
Closing due to stainless. Please feel free to reopen the issue if it is still a problem.

disperso, 2018-11-06:
> @jgramsch <https://github.com/jgramsch> @disperso
> <https://github.com/disperso> Sorry to bump a 1 year old issue, but is
> this bug still prevalent?

I'm sorry, I didn't see your message till now. I have been a little out of
this for a while (sadly) but I think I just started using arduino IDE with
an external editor and it worked fine.

Regards,

  Javier

StefanMeGit, 2018-11-11:
Same for me... have an Feather M0 with SAMD21 and it says |no Arduino connected|

Ardunio IDE says:

```
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Mar  5 2016 17:46:52
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.851 seconds

Write 74104 bytes to flash (1158 pages)

[=                             ] 5% (64/1158 pages)
[===                           ] 11% (128/1158 pages)
[====                          ] 16% (192/1158 pages)
[======                        ] 22% (256/1158 pages)
[========                      ] 27% (320/1158 pages)
[=========                     ] 33% (384/1158 pages)
[===========                   ] 38% (448/1158 pages)
[=============                 ] 44% (512/1158 pages)
[==============                ] 49% (576/1158 pages)
[================              ] 55% (640/1158 pages)
[==================            ] 60% (704/1158 pages)
[===================           ] 66% (768/1158 pages)
[=====================         ] 71% (832/1158 pages)
[=======================       ] 77% (896/1158 pages)
[========================      ] 82% (960/1158 pages)
[==========================    ] 88% (1024/1158 pages)
[============================  ] 93% (1088/1158 pages)
[============================= ] 99% (1152/1158 pages)
[==============================] 100% (1158/1158 pages)
done in 0.472 seconds

Verify 74104 bytes of flash with checksum.
Verify successful
done in 0.052 seconds
CPU reset.
```

203Null, 2018-12-24:
Having the same issus with stm32duino, which dont have a com port, upload with dfu.

floraledcompany, 2019-04-03:
Adding my voice to this bug,  Feather M0 with SAMD21 here, "No arduino found!" Everytime I try and upload or use the serial monitor. Build and Verify work fine, and they report the correct amount of memory the sketch uses for the board.