Sorunome/arduino-upload #25: error : File isn't part of an Arduino sketch!

rafaelmmoraes, 2017-08-16:
Hey,

Can you help me? I got this error when I try to verify.
"File isn't part of an Arduino sketch!"

thanks

Sorunome, 2017-08-16:
that means that the currently selected file isn't part of an arduino sketch.

For example, if your main file is called `sketch.ino` that mean that that file `sketch.ino` must be in a folder called `sketch`.

Or, for example `game.ino` must be in a folder called `game`

I hope that helps you out ^.^

rafaelmmoraes, 2017-08-16:
OKay, but now i have other msg..."'arduino' could not be spawned. Is it installed and on your path? If so please open an issue on the package spawning the process."

Sorunome, 2017-08-16:
Which OS do you use?

Which version of the Arduino IDE do you have installed?

rafaelmmoraes, 2017-08-16:
I'm using OS Sierra.

Arduino IDE? it's 1.8.3, but why do i need arduino IDE if i'm using Atom?

Sorunome, 2017-08-16:
Sierra is....OSX?

This atom plugin invokes the arduino IDE via command line options in a way that it'll never open the gui of it: it needs the arduino ide for compiling and uploading sketches.

As for the issue of `'arduino' could not be spawned.` that basically means that this plugin can't find the arduino IDE. In the settings of arduino-upload you'll see an option `Arduino Executable Path`, enter there the full path to where the arduino executable is. so like `/path/to/arduino`. (Perhaps `/Applications/Arduino.app/Contents/MacOS/Arduino` ?)

If you can't figure that out, i'm sorry I can't help you with that, i don't have any apple machine. However, there is already areport on that open, over here: https://github.com/Sorunome/arduino-upload/issues/13

Sorunome, 2017-08-16:
And here, too: https://github.com/Sorunome/arduino-upload/issues/4

rafaelmmoraes, 2017-08-16:
That's great. thanks a lot!

Sorunome, 2017-08-16:
I think the problem is resolved now, if it isn't then feel free to re-open

closed 2017-08-16 by Sorunome