Sorunome/arduino-upload #27: Cannot install serialport & usb-detection dependencies

mrwunderbar666, 2017-08-31:
Running Atom on Windows 10 and I installed "arduino-upload" via the Atom GUI and also tried via APM.

In neither cases it installs the npm dependencies for the serial port and usb detection.

How to do that? I need help! :)

Sorunome, 2017-08-31:
You could try to manually install serialport via `npm install serialport`. Other than that, i have no idea. See https://github.com/Sorunome/arduino-upload/issues/8 for more information, I'd be very happy for help on this issue.

Sorunome, 2017-09-01:
I did a lot of digging today, it should work now!

closed 2017-09-01 by Sorunome

mrwunderbar666, 2017-09-01:
Thanks,
I'll give it a shot tomorrow!

I've tried installing the dependencies manually yesterday with `npm` but was no success.
I got the `serialport` package somewhat working, but `usb-detect` just wouldn't work under windows 10.
Let's see, and thanks.

Sorunome, 2017-09-01:
`usb-detect` is only needed if you use a programmer, such as atmel ice