Sorunome/arduino-upload #28: The package breaks on every update

qub1750ul, 2017-08-31:
As the title, then Atom is able to fix the problem on its own by offering the possibility of auto-rebuilding the package internally.
This problem is present in more than one version of the package.

Maybe related to #27 ?

Atom 1.19.4 x64
OS: Arch Linux ( last full package update today )

~~~
Some installed packages could not be loaded because they contain native modules that were compiled for an earlier version of Atom.

arduino-upload 1.2.0

build@unknown – Cannot find module '/home/root/.atom/packages/arduino-upload/node_modules/serialport/build'
serialport@5.0.0 – The module '/home/root/.atom/packages/arduino-upload/node_modules/serialport/build/Release/serialport.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 50. This version of Node.js requires NODE_MODULE_VERSION 53. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or`npm install`).
usb-detection@1.4.0 – The module '/home/root/.atom/packages/arduino-upload/node_modules/usb-detection/build/Release/detection.node' was compiled against a different Node.js version using NODE_MODULE_VERSION 50. This version of Node.js requires NODE_MODULE_VERSION 53. Please try re-compiling or re-installing the module (for instance, using `npm rebuild` or`npm install`).
build@1.4.0 – Cannot find module '/home/root/.atom/packages/arduino-upload/node_modules/usb-detection/build'
~~~

Sorunome, 2017-08-31:
> then Atom is able to fix the problem on its own by offering the possibility of auto-rebuilding the package internally.

So just re-build? It's a click of a button?

> Maybe related to #27 ? 

Perhaps, because serialport uses node-gyp and that is still funky with electron. With windows it doesn't work at all, with linux it at least works......but stripping that would mean removing all the serialport features. Or do you know of any other node serialport package that doesn't suffer of these issues?

qub1750ul, 2017-08-31:
> So just re-build? It's a click of a button?

Yes

> Perhaps, because serialport uses node-gyp and that is still funky with electron. With windows it doesn't work at all, with linux it at least works......but stripping that would mean removing all the [...]

I'm not practical of node.

Sorunome, 2017-09-01:
As I provide builds for serialport now, does it maybe work?

qub1750ul, 2017-09-01:
Reinstalling the package on a clean Atom install, the problem persists

StaticRocket, 2018-04-01:
[image]
Visual Example