Sorunome/arduino-upload #30: Cant install package on OSX

cr4nk89, 2017-10-29:
Always getting this fault when I try to install the package. 

`Installing “arduino-upload@1.3.2” failed.Hide output…

> serialport-builds-electron@4.0.7 install /private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron
> node-pre-gyp install --fallback-to-build

Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release' (1)
/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9
`-- (empty)

node-pre-gyp ERR! Tried to download(500): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-darwin-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport-builds-electron@4.0.7 and electron@1.6.15 (electron-v1.6 ABI) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: Python executable "/Applications/Atom.app/Contents/Resources/app/apm/bin/python-interceptor.sh" is v3.6.1, which is not supported by gyp.
gyp ERR! stack You can pass the --python switch to point to Python >= v2.5.0 & < 3.0.0.
gyp ERR! stack     at failPythonVersion (/Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/lib/configure.js:454:14)
gyp ERR! stack     at /Applications/Atom.app/Contents/Resources/app/apm/node_modules/node-gyp/lib/configure.js:443:9
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:197:7)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
gyp ERR! System Darwin 16.7.0
gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/.bin/node-gyp" "configure" "--fallback-to-build" "--module=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node" "--module_name=serialport" "--module_path=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release"
gyp ERR! cwd /private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/Applications/Atom.app/Contents/Resources/app/apm/bin/node /Applications/Atom.app/Contents/Resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node --module_name=serialport --module_path=/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Darwin 16.7.0
node-pre-gyp ERR! command "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/node_modules/arduino-upload/node_modules/serialport-builds-electron
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.38
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/package.json'
npm WARN apm-install-dir-117929-77429-uih4nw.272gdjkyb9 No description
npm WARN apm-install-dir-117929-77429-uih4nw.272gdjkyb9 No repository field.
npm WARN apm-install-dir-117929-77429-uih4nw.272gdjkyb9 No README data
npm WARN apm-install-dir-117929-77429-uih4nw.272gdjkyb9 No license field.
npm ERR! Darwin 16.7.0
npm ERR! argv "/Applications/Atom.app/Contents/Resources/app/apm/bin/node" "/Applications/Atom.app/Contents/Resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/Users/User/.atom/.apm/.apmrc" "--userconfig" "/Users/User/.atom/.apmrc" "install" "/private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/d-117929-77429-3py533.xubv9m5cdi/package.tgz" "--runtime=electron" "--target=1.6.15" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport-builds-electron@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport-builds-electron@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport-builds-electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport-builds-electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport-builds-electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /private/var/folders/0y/4dl944691xxgnd8csdhp_ksm0000gn/T/apm-install-dir-117929-77429-uih4nw.272gdjkyb9/npm-debug.log
npm ERR! code 1
`

Sorunome, 2017-10-31:
wha, um, i will have to look into that

vlsd, 2018-02-27:
it seems like the server hosting the file  https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-darwin-x64.tar.gz doesn't work any longer; that file (and newer versions thereof) can be found here  https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-darwin-x64.tar.gz

Sorunome, 2018-02-27:
i don't see any difference between the two URLs you posted?

vlsd, 2018-02-27:
copy paste error, sorry; edited

Sorunome, 2018-02-27:
The issue with the releases is, that when i tried newer serialport versions stuff simply.......didn't work, which is why i started to host pre-builds.

Guess I'll have to look into this again

vlsd, 2018-02-27:
Yeah, I figured that's why you pointed to an older version. I'm not even sure if the server being down is the main error or just ancillary, tbh. I just started attacking the first error I saw :)

Sorunome, 2018-02-27:
nah, for some reason it gives error 500 instead of 404 if the file isn't found (prolly some mis-config), it's just that the file isn't present

PR #58 (2018-10-10): Serialport installed and everything updated
- Pushed all versions to latest and migrated to the official serialport package instead of your custom one.
- Made usb-detection a standard dependency (it's obviously required).
- Added package-lock.json to the .gitignore so it doesn't interfere with `apm install`. Also removed screenshots from it because I had made a mistake earlier.
- Fixed my troubleshooting section

Confirmed working on Windows 10 and the latest version of Arch Linux.

This closes #55, closes #52, closes #43, closes #39, closes #41, closes #32, and closes #30.

This patch also ma [... cut]
closed 2018-10-10 by Sorunome