Sorunome/arduino-upload #34: package opens IDE

SpeedrunnerG55, 2018-02-01:
whenever I start atom with the package it will also open the IDE.
when I verify, build or upload it will print the message "Start building..." then open another IDE window each time with my code in it and then does nothing. 

whenever I upload
it will also open a window with the title "bad file selected" with the message 
"Processing can only open its own sketches and other files ending in .ino or .pde" 

closing the IDE window causes atom to become unresponsive.

OS: Ubuntu 16.04
using atom 1.22.0
arduino-upload 1.3.2

Sorunome, 2018-02-01:
which version of the arduino IDE are you on?

SpeedrunnerG55, 2018-02-01:
2:1.05+dfsg2-4

Sorunome, 2018-02-01:
uuuh what does it say if you open the arduino IDE, go to `Help` -> `About Arduino` small in the top-left corner as version?

SpeedrunnerG55, 2018-02-01:
that's what it says, "2:1.05+dfsg2-4"

Sorunome, 2018-02-01:
hmm, that seems to be based on `1.0.5`

This package only really works with IDE `1.6.x` and above, so you'd have to manually install & link the executable in the config, you can download it from the arduino website https://www.arduino.cc/en/Main/Software

SpeedrunnerG55, 2018-02-01:
seems to be working now thanks for the help.

Sorunome, 2018-02-01:
Any time!

closed 2018-02-01 by Sorunome

invadrew, 2018-11-22:
I have same issue even so I've installed latest Arduino IDE (1.8.5), and whenever I try to verify or upload sketch new Arduino IDE window appears, furthermore, I have to manually press the button in Arduino IDE to verify, upload etc. selected sketch. Can it be somehow connected that I'm using xfce4 desktop environment.

OS: Xubuntu 18.04
using atom 1.31.0
arduino-upload 1.4.1

invadrew, 2018-11-23:
I've just solve this problem. The problem was with ardino command line executable because in PATH was specified bash script with following code:
```
#!/bin/bash

"/home/andrew/.local/share/arduino-1.8.7/arduino/arduino"
```
I replace this file with soft link to launcher, so now everything works fine.