Sorunome/arduino-upload #36: Manual Port Selection

StaticRocket, 2018-03-08:
Proposing a second drop-up list for open ports next to the board selection drop-up list. Could be potentially helpful for troubleshooting errors like #24 and allow for multiple connected boards without upload issues.

shawnhochstetler, 2018-11-21:
Would really love to see this as well, I'm trying to develop for the Adafruit ESP32 Feather and can't upload due to the board not being recognized on the serial port. These boards use the Silicon Labs CP210x USB to UART Bridge drivers and show up as exactly that in the device manager.

Sorunome, 2018-11-21:
Do you happen to have their vendor / product ID? If you are on linux, they typically show up in `lsusb`

shawnhochstetler, 2018-11-21:
I'm on windows 10, where can I find that info for you?

Sorunome, 2018-11-21:
does this help any? https://superuser.com/questions/1106247/how-can-i-get-the-vendor-id-and-product-id-for-a-usb-device/1106248

shawnhochstetler, 2018-11-21:
[image]
Sorry, I should have just googled how to find that.

Sorunome, 2018-11-21:
added that device in https://github.com/Sorunome/arduino-upload/commit/31876c748e1ef062db192a5e6571da745595ab71 now as a quick-fix (should also be published to apm, so that you get an update), OFC the original issue is still unsovled