Sorunome/arduino-upload #37: Option to automatically save file before upload please

stibinator, 2018-03-11:
I can't count the number of times I've uploaded something and spent ages trying to work out why the changes I've made to the code didn't do anything. Then I realise that I didn't save before the last upload.

Could you please implement at least an option to automatically save before uploading, to make your fine package even finer. Thanks

StaticRocket, 2018-03-14:
I have a PR open ( #38 ) that makes "save active tabs" a toggleable setting that is disabled by default. It closes the serial monitor and saves all active tabs before building/uploading/verifying. Is there anything else you think I should address with this system?

stibinator, 2018-03-14:
Sounds like a plan. Defaulting to the current behaviour is probably the best way to introduce it. 

Does the serial monitor automatically re-open? It would be annoying if you want to say, monitor how your script initialises and you had to manually re-open the serial monitor after uploading.

StaticRocket, 2018-03-14:
I added a toggle to specify whether or not to reopen the serial monitor if it was open previously and Autosave is enabled as well.

PR #38 (2018-03-21): Save all active tabs before build and upload
Proposing the change to run window:save-all command before beginning the upload or building process. Unfortunately this needs to close the serial monitor to avoid saving its output in the project directory. Resolves #37
closed 2018-03-21 by Sorunome