Sorunome/arduino-upload #39: Unable to install on Linux

nottelmo, 2018-03-12:
´´´/bin/bash: module: line 1: syntax error: unexpected end of file
/bin/bash: error importing function definition for `module'
/bin/bash: scl: line 1: syntax error: unexpected end of file
/bin/bash: error importing function definition for `scl'

> serialport-builds-electron@4.0.7 install /tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron
> node-pre-gyp install --fallback-to-build

Failed to execute '/usr/share/atom/resources/app/apm/bin/node /usr/share/atom/resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node --module_name=serialport --module_path=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release --python=/usr/share/atom/resources/app/apm/bin/python-interceptor.sh' (1)
/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi
└── (empty)

sh: scl: line 1: syntax error: unexpected end of file
sh: error importing function definition for `scl'
sh: module: line 1: syntax error: unexpected end of file
sh: error importing function definition for `module'
node-pre-gyp ERR! Tried to download(500): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.6-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport-builds-electron@4.0.7 and electron@1.6.16 (electron-v1.6 ABI, glibc) (falling back to source compile with node-gyp) 
/bin/bash: scl: line 1: syntax error: unexpected end of file
/bin/bash: error importing function definition for `scl'
/bin/bash: module: line 1: syntax error: unexpected end of file
/bin/bash: error importing function definition for `module'
/bin/sh: scl: line 1: syntax error: unexpected end of file
/bin/sh: error importing function definition for `scl'
/bin/sh: module: line 1: syntax error: unexpected end of file
/bin/sh: error importing function definition for `module'
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/atom/resources/app/apm/node_modules/node-gyp/lib/configure.js:305:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.15.7-300.fc27.x86_64
gyp ERR! command "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/.bin/node-gyp" "configure" "--fallback-to-build" "--module=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node" "--module_name=serialport" "--module_path=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release" "--python=/usr/share/atom/resources/app/apm/bin/python-interceptor.sh"
gyp ERR! cwd /tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/share/atom/resources/app/apm/bin/node /usr/share/atom/resources/app/apm/bin/../node_modules/.bin/node-gyp configure --fallback-to-build --module=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release/serialport.node --module_name=serialport --module_path=/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron/build/Release --python=/usr/share/atom/resources/app/apm/bin/python-interceptor.sh' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.15.7-300.fc27.x86_64
node-pre-gyp ERR! command "/usr/share/atom/resources/app/apm/bin/node" "/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/node_modules/arduino-upload/node_modules/serialport-builds-electron
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/package.json'
npm WARN apm-install-dir-118212-4160-1bor4wn.dp845tgldi No description
npm WARN apm-install-dir-118212-4160-1bor4wn.dp845tgldi No repository field.
npm WARN apm-install-dir-118212-4160-1bor4wn.dp845tgldi No README data
npm WARN apm-install-dir-118212-4160-1bor4wn.dp845tgldi No license field.
npm ERR! Linux 4.15.7-300.fc27.x86_64
npm ERR! argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/not-telmo/.atom/.apm/.apmrc" "--userconfig" "/home/not-telmo/.atom/.apmrc" "install" "/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz" "--runtime=electron" "--target=1.6.16" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport-builds-electron@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport-builds-electron@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport-builds-electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport-builds-electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport-builds-electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /tmp/apm-install-dir-118212-4160-1bor4wn.dp845tgldi/npm-debug.log
npm ERR! code 1´´´

StaticRocket, 2018-03-12:
I've heard a lot of similar [issues](https://deftly.net/posts/2017-06-01-measuring-the-weight-of-an-electron.html) with installing Electron in the community, I'm not sure if this could help but [here](https://electronjs.org/docs/development/build-instructions-linux) is the official Electron build instructions for linux.

StaticRocket, 2018-03-16:
Upon further investigation it is not an issue with electron, [this thread](https://github.com/AtomLinter/linter-puppet-lint/issues/68) indicates that it is instead some issue with `environment-modules` for bash in Fedora. One of the last two solutions mentioned should fix this for you.

nottelmo, 2018-03-18:
@StaticRocket Did the last two solutions on that thread but it is still not working

StaticRocket, 2018-03-18:
Run this command:
```
"/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/not-telmo/.atom/.apm/.apmrc" "--userconfig" "/home/not-telmo/.atom/.apmrc" "install" "/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz" "--runtime=electron" "--target=1.6.1" "--arch=x64" "--global-style"
```
The install was pointed to dead links and invalid fallback version was set. This was the same issue I had on #40.

nottelmo, 2018-03-19:
Thanks for you efforts, but it is still not working. Below is the resulting log:

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/share/atom/resources/app/apm/bin/node',
1 verbose cli   '/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js',
1 verbose cli   '--globalconfig',
1 verbose cli   '/home/not-telmo/.atom/.apm/.apmrc',
1 verbose cli   '--userconfig',
1 verbose cli   '/home/not-telmo/.atom/.apmrc',
1 verbose cli   'install',
1 verbose cli   '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz',
1 verbose cli   '--runtime=electron',
1 verbose cli   '--target=1.6.1',
1 verbose cli   '--arch=x64',
1 verbose cli   '--global-style' ]
2 info using npm@3.10.10
3 info using node@v6.9.5
4 silly loadCurrentTree Starting
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 silly fetchPackageMetaData /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz
8 silly fetchOtherPackageData /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz
9 silly cache add args [ '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz', null ]
10 verbose cache add spec /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz
11 silly cache add parsed spec Result {
11 silly cache add   raw: '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz',
11 silly cache add   scope: null,
11 silly cache add   escapedName: null,
11 silly cache add   name: null,
11 silly cache add   rawSpec: '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz',
11 silly cache add   spec: '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz',
11 silly cache add   type: 'local' }
12 error addLocal Could not install /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz
13 silly fetchPackageMetaData Error: ENOENT: no such file or directory, open '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz'
13 silly fetchPackageMetaData     at Error (native)
13 silly fetchPackageMetaData  error for /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz { Error: ENOENT: no such file or directory, open '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz'
13 silly fetchPackageMetaData     at Error (native)
13 silly fetchPackageMetaData   errno: -2,
13 silly fetchPackageMetaData   code: 'ENOENT',
13 silly fetchPackageMetaData   syscall: 'open',
13 silly fetchPackageMetaData   path: '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz' }
14 silly rollbackFailedOptional Starting
15 silly rollbackFailedOptional Finishing
16 silly runTopLevelLifecycles Finishing
17 silly install printInstalled
18 verbose stack Error: ENOENT: no such file or directory, open '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz'
18 verbose stack     at Error (native)
19 verbose cwd /home/not-telmo
20 error Linux 4.15.9-300.fc27.x86_64
21 error argv "/usr/share/atom/resources/app/apm/bin/node" "/usr/share/atom/resources/app/apm/node_modules/npm/bin/npm-cli.js" "--globalconfig" "/home/not-telmo/.atom/.apm/.apmrc" "--userconfig" "/home/not-telmo/.atom/.apmrc" "install" "/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz" "--runtime=electron" "--target=1.6.1" "--arch=x64" "--global-style"
22 error node v6.9.5
23 error npm  v3.10.10
24 error path /tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz
25 error code ENOENT
26 error errno -2
27 error syscall open
28 error enoent ENOENT: no such file or directory, open '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz'
29 error enoent ENOENT: no such file or directory, open '/tmp/d-118212-4160-izydi7.6uc1o9wwmi/package.tgz'
29 error enoent This is most likely not a problem with npm itself
29 error enoent and is related to npm not being able to find a file.
30 verbose exit [ -2, true ]
```

StaticRocket, 2018-03-19:
Ah, considering the amount of time that has pass between the two commands you need to uninstall whatever was saved by using `apm remove arduino-upload` and then run `apm install arduino-upload` to get back to where you were.
(Your temp file for the install was removed so you need to make another one using the steps mentioned previously.)

nottelmo, 2018-03-19:
Still nothing,
`apm remove arduino-upload` says package not found,

`apm install arduino-upload` still gets us back to the starting point of this post,

The command you provided still outputs the same log from my previous comment.

As for the file in the tmp/ folder, it was never created in any of the steps.

StaticRocket, 2018-03-23:
Have you tried it since #40 was closed?

nottelmo, 2018-03-24:
@StaticRocket Assuming the package found in Atom is the latest one (1.3.3), the same problem still persists.

StaticRocket, 2018-03-24:
Just out of curiosity, what version of Atom are you running?

nottelmo, 2018-03-24:
Just updated to the latest version (1.25) and the problem persists.

StaticRocket, 2018-03-26:
Is the log output from running `apm install arduino-upload` with the either "fix" from that thread I linked the same as the original log, or did we at least get something different?

edward-p, 2018-07-23:
It seems that is a download failure.
My log shows:

```
node-pre-gyp ERR! Tried to download(500): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v2.0-linux-x64.tar.gz 
````

StaticRocket, 2018-07-23:
@edward-p That totally seems like a problem, right? Well... [miraculously it isn't](https://github.com/Sorunome/arduino-upload/issues/46#issuecomment-405119826). At least, not at the moment.

Sorunome, 2018-07-23:
If node-pre-gyp fails it falls back to node-gyp which compiles packages itself. That requires build tools to be present, which should be sstandard in a linux install

edward-p, 2018-07-23:
@StaticRocket Yes, it is like [#40](https://github.com/Sorunome/arduino-upload/issues/40). I think my electron version is too high.

StaticRocket, 2018-07-23:
@Sorunome I remember reading in a few logs that the fallback version was never set and is not being passed to npm. Normally this just threw up a few errors and continued the install, but it may be worth checking out.

PR #58 (2018-10-10): Serialport installed and everything updated
- Pushed all versions to latest and migrated to the official serialport package instead of your custom one.
- Made usb-detection a standard dependency (it's obviously required).
- Added package-lock.json to the .gitignore so it doesn't interfere with `apm install`. Also removed screenshots from it because I had made a mistake earlier.
- Fixed my troubleshooting section

Confirmed working on Windows 10 and the latest version of Arch Linux.

This closes #55, closes #52, closes #43, closes #39, closes #41, closes #32, and closes #30.

This patch also ma [... cut]
closed 2018-10-10 by Sorunome