Sorunome/arduino-upload #40: Dead prebuilt binaries

StaticRocket, 2018-03-16:
Figured I might as well try out `apm dev <package-name>` seeing as how often I have been picking through the files for this project and discovered that apm is unable to build this project. Further investigation points to `node-pre-gpy` trying to load a prebuilt binaries for electron that were previously hosted through your website.
```
node-pre-gyp ERR! Tried to download(500): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v1.7-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport-builds-electron@4.0.7 and electron@1.7.11 (electron-v1.7 ABI, unknown) (falling back to source compile with node-gyp)
```
That link is now dead and `node-pre-gpy` is falling back on another invalid link as evident by `gyp ERR! stack Error: 404 response downloading https://iojs.org/download/release/v1.7.11/iojs-v1.7.11.tar.gz`. Did you mean to reference `https://iojs.org/download/release/v1.7.1/iojs-v1.7.1.tar.gz` instead? I would submit a pull request with a fix to the latter, if that's the case, but I don't know where these values are defined within the package.

StaticRocket, 2018-03-16:
The latter issue was actually due to an invalid fallback version.

Sorunome, 2018-03-19:
Oh, atom updated to electron 1.7, i only served up to electron 1.6.....time to rebuild! :D

Sorunome, 2018-03-19:
aaaand should be done. Please re-open if the issue persists

closed 2018-03-19 by Sorunome

peterPAN85-ofthesun, 2021-12-15:
Hello I have the same issue.
I have an Archlinux System so all my software and libraries are up to date. Do you think electron is in most younger version to use with Atom again?