Sorunome/arduino-upload #41: Another Install Issue on Mac OS

sww1235, 2018-03-30:
Hello:

I recently updated to Atom 1.25.0 and along with that attempted to update to Arduino upload 1.33.

The build failed with the log [here](https://gist.github.com/sww1235/6963607465a5092529cfd6ca0cdda4f4)

It looks like it is failing to build due to it requiring a newer version of npm. 

related to #30 and #32

jrmedd, 2018-05-24:
I'm in the same boat on this. [My log, for the record](https://gist.github.com/jrmedd/6c17e4c3b760e8c0a7c75928836e8ad9).

StaticRocket, 2018-05-26:
`-- (empty)` indicates the issues reported in #40 are still present. The fall back version is still not set.

qoda, 2018-05-29:
`sudo npm install -g npm` seemed to do the trick on OSX for me.

jrmedd, 2018-05-29:
@qoda Can you give any more specific details?

jrmedd, 2018-06-05:
I realised that my /usr/local/bin/python was symlinked to my Homebrew install of Python 3. I ran brew unlink python and install worked!

PR #58 (2018-10-10): Serialport installed and everything updated
- Pushed all versions to latest and migrated to the official serialport package instead of your custom one.
- Made usb-detection a standard dependency (it's obviously required).
- Added package-lock.json to the .gitignore so it doesn't interfere with `apm install`. Also removed screenshots from it because I had made a mistake earlier.
- Fixed my troubleshooting section

Confirmed working on Windows 10 and the latest version of Arch Linux.

This closes #55, closes #52, closes #43, closes #39, closes #41, closes #32, and closes #30.

This patch also ma [... cut]
closed 2018-10-10 by Sorunome