Sorunome/arduino-upload #43: Can't open Serial monitor

FlorianProgrammiert, 2018-05-11:
I can't open the Serial monitor. The only Error i get is: "Serialport dependency not present, try installing it! (And, if you figure out how, please report me how here as I don't know how to do it..... Really, please help me! D: )". I've tried installing Visual c++ building tools 2015 and Visual c++ building tools 2017. Nothing helped.
I am using Atom 1.26.1 x64 at Windows 10 x64 (and arduino IDE 1.6.8)
Can anybody help me?

Bye,
Florian

wesley-cantarino, 2018-05-16:
I too. I don't understand this problem

stibinator, 2018-05-18:
I'm sure the serial monitor used to work…?

CodeJunkieio, 2018-06-26:
@FlorianProgrammiert @wesley-cantarino @stibinator This error isn't an error. The developer couldn't get the serial monitor to work, He's telling you to try it for yourself and if you have any luck, let him know so he can fix it.

Sorunome, 2018-06-26:
what works for me occasionally is uninstalling and re-installing the package.

Also, i really need to move this to the newer serialport versions, if anyone wants to help, that'd be great

PR #58 (2018-10-10): Serialport installed and everything updated
- Pushed all versions to latest and migrated to the official serialport package instead of your custom one.
- Made usb-detection a standard dependency (it's obviously required).
- Added package-lock.json to the .gitignore so it doesn't interfere with `apm install`. Also removed screenshots from it because I had made a mistake earlier.
- Fixed my troubleshooting section

Confirmed working on Windows 10 and the latest version of Arch Linux.

This closes #55, closes #52, closes #43, closes #39, closes #41, closes #32, and closes #30.

This patch also ma [... cut]
closed 2018-10-10 by Sorunome