Sorunome/arduino-upload #44: Add command to Upload with Programmer

0xdec, 2018-05-30:
It would be great if we could upload using an external programmer, like in the Arduino IDE. I can see in `arduino-upload.coffee` that you have already implemented some aspects of port selection. I would change it myself but my CoffeeScript knowledge is pretty rusty and I can't for the life of me figure out how to add another command that sets the port to `'PROGRAMMER'`.

Sorunome, 2018-05-31:
It does that already automatically if a programmer is detected, tested with an Atmel ICE (I don't have USB manufacturer/product IDs of the other)

0xdec, 2018-05-31:
For some reason it doesn't work for me. I think the -useprogrammer flag might not be passed to arduino?
OS: Arch Linux
Board: Arduino Uno
Programmer: Atmel Corp AVRISP mkII (USB VID:PID 03eb:2104)
Port: usb