Sorunome/arduino-upload #46: Installing failed

sergerold, 2018-07-06:
Installing “arduino-upload@1.3.3” failed.Hide output…

> serialport-builds-electron@4.0.7 install C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron
> node-pre-gyp install --fallback-to-build

Failed to execute 'C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\bin\node.exe C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release' (1)
C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr
`-- (empty)

node-pre-gyp ERR! Tried to download(500): https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v2.0-win32-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport-builds-electron@4.0.7 and electron@2.0.4 (electron-v2.0 ABI, unknown) (falling back to source compile with node-gyp) 
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\node_modules\node-gyp\lib\configure.js:449:14)
gyp ERR! stack     at C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\node_modules\node-gyp\lib\configure.js:404:11
gyp ERR! stack     at C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:123:15)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Users\\tom\\AppData\\Local\\atom\\app-1.28.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\tom\\AppData\\Local\\atom\\app-1.28.1\\resources\\app\\apm\\node_modules\\node-gyp\\bin\\node-gyp.js" "configure" "--fallback-to-build" "--module=C:\\Users\\tom\\AppData\\Local\\Temp\\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\\node_modules\\arduino-upload\\node_modules\\serialport-builds-electron\\build\\Release\\serialport.node" "--module_name=serialport" "--module_path=C:\\Users\\tom\\AppData\\Local\\Temp\\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\\node_modules\\arduino-upload\\node_modules\\serialport-builds-electron\\build\\Release"
gyp ERR! cwd C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron
gyp ERR! node -v v6.9.5
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\bin\node.exe C:\Users\tom\AppData\Local\atom\app-1.28.1\resources\app\apm\bin\\..\node_modules\node-gyp\bin\node-gyp.js configure --fallback-to-build --module=C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release\serialport.node --module_name=serialport --module_path=C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron\build\Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\node-pre-gyp\lib\util\compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:877:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Windows_NT 10.0.17134
node-pre-gyp ERR! command "C:\\Users\\tom\\AppData\\Local\\atom\\app-1.28.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\tom\\AppData\\Local\\Temp\\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\\node_modules\\arduino-upload\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\node_modules\arduino-upload\node_modules\serialport-builds-electron
node-pre-gyp ERR! node -v v6.9.5
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\package.json'
npm WARN apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr No description
npm WARN apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr No repository field.
npm WARN apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr No README data
npm WARN apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr No license field.
npm ERR! Windows_NT 10.0.17134
npm ERR! argv "C:\\Users\\tom\\AppData\\Local\\atom\\app-1.28.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\tom\\AppData\\Local\\atom\\app-1.28.1\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\tom\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\tom\\.atom\\.apmrc" "install" "C:\\Users\\tom\\AppData\\Local\\Temp\\d-11866-22924-oti06s.srls68ncdi\\package.tgz" "--runtime=electron" "--target=2.0.4" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE

npm ERR! serialport-builds-electron@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the serialport-builds-electron@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the serialport-builds-electron package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs serialport-builds-electron
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls serialport-builds-electron
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\tom\AppData\Local\Temp\apm-install-dir-11866-22924-w4fcsr.tbkuyxpqfr\npm-debug.log
npm ERR! code 1

sergerold, 2018-07-15:
Is the issue that the following file is unavailable: https://files.sorunome.de/arduino-upload/serialport/4.0.7/serialport-v4.0.7-electron-v2.0-win32-x64.tar.gz

StaticRocket, 2018-07-17:
@sergerold If it is, then it looks like it's a repeat of #40.

Sorunome, 2018-07-17:
which should be fixed when updating to latest serialport dependency, but, you know, time...

StaticRocket, 2018-07-17:
@sergerold I was unable to replicate this. The only thing I can think of that would be different between our apm instiances would be build tools. Did you properly config apm for your msvs version and point it to python with `apm config set msvs_version [your-year-here]` and `apm config set python [your-path]\python.exe`? I don't know why that would change anything, but it's the only thing I can think of.

StaticRocket, 2018-07-17:
@rodrigocaus I ended up having to do a similar thing but I skipped the Node.js installation using [this](https://github.com/lgeiger/windows-build-tools) package. It's a bit fidgety and still requires some configuration but if anyone else needs to do this and doesn't wish to have 2 Node installs I gave a little bit of [insight](https://github.com/lgeiger/windows-build-tools/issues/4) on that method.

sergerold, 2018-07-17:
@StaticRocket I don't use MSVS and don't know how to use it. I didn't think I needed it to install the arduino-upload package for atom - I recall using this package a while ago without using MSVS. Are there any instructions on what I need to do with MSVS?

Thanks

StaticRocket, 2018-07-17:
@sergerold If you followed rodrigocaus' instructions MSVS 2015 should have been installed. npm was automatically configured but apm may not have been. Simply send the command `apm config set msvs_version 2015` and you should be good for MSVS. Then just be sure to config apm for python like stated above. apm acts as a wrapper for npm and if you do an isolated npm install it may need to be configured for apm, so you should expect to need to do the same for python.