Sorunome/arduino-upload #47: Verify/build/upload output not visible in Atom

lorenzos, 2018-07-07:
The package works great, but I can't see the output of any command, i.e. the bottom panel shown in the first GIF in README is never shown. The only way I can see it is if I open the Electron developer tools (Ctrl+Shift+I), but it's a mess and it's not clickable to jump to errors. I think that panel is something related to some `atom-space-pen-views` thing, but I don't know if I have to install some other package to make it work, or what.

Tested in Atom 1.26.1 and 1.28.1, same issue in both.

Thanks.

Sorunome, 2018-07-07:
Do you perhaps have your language set to something else than english? As then the current parsing of arduino output (of where is build output and where upload output) will fail, and might result in this

lorenzos, 2018-07-07:
Yes! It works now, thank you!

closed 2018-07-07 by lorenzos