Sorunome/arduino-upload #52: Windows Build Tools Dependency

StaticRocket, 2018-07-18:
Why does this seem to require windows build tools for the initial install? I don't believe it has anything to do with Python and the only way I see it needing MSVS is if something was being compiled during install. Obviously there must be some way to work around this or users would have reported a similar experience with other Atom packages. Submitting this issue for idea spitballing.

Sorunome, 2018-07-18:
> Why does this seem to require windows build tools for the initial install?

Because of the serialport dependency version this is using (4.0.7, was newest at the time of writing) only provided pre-builts for node.js runtime, not for electron runtime.

The fix would probably be to update to the (by now) newest serialport version, which also provides electron builds. That, unfortunately, takes time I currently don't have. I'd be happy for anyone wanting to help.

StaticRocket, 2018-07-18:
I'd offer to help, but I have little to no experience with Electron. I wouldn't want to get in the way.

Sorunome, 2018-07-18:
That shouldn't be a problem, apm does all that stuff for you. So it's only about migrating serialport to newest version (incompatiple APIs)

StaticRocket, 2018-07-18:
@Sorunome Well... I'm still working on figuring out where to start with the serialport-builds-electron upgrade, but I think I've solved the serialport installation issue you mentioned in that one error message. I installed it by listing it as a package dependency and sending `Update Package Dependencies: Update` through Atom's Command Palette while in dev mode for the package.

Sorunome, 2018-07-18:
You'll need the newest serialport package. serialport-builds-electron is my own fork for literally this atom package only, which only provides some more pre-builts.

PR #58 (2018-10-10): Serialport installed and everything updated
- Pushed all versions to latest and migrated to the official serialport package instead of your custom one.
- Made usb-detection a standard dependency (it's obviously required).
- Added package-lock.json to the .gitignore so it doesn't interfere with `apm install`. Also removed screenshots from it because I had made a mistake earlier.
- Fixed my troubleshooting section

Confirmed working on Windows 10 and the latest version of Arch Linux.

This closes #55, closes #52, closes #43, closes #39, closes #41, closes #32, and closes #30.

This patch also ma [... cut]
closed 2018-10-10 by Sorunome