Sorunome/arduino-upload #56: Main functions not defined

alexdrymonitis, 2018-08-19:
I installed the package on Ubunstu Studio 18.04, Atom 1.29.0.
Trying to upload a test blink sketch I get:
```
undefined reference to 'setup'
undefined reference to 'loop'
```
I've set the path to my Arduino executable, which is 1.8.5
I also see that it's using an older version of the Arduino, 1.6.21, right? Can that be an issue?

Sorunome, 2018-08-19:
what is the arduino sketch you are trying to compile & upload?

alexdrymonitis, 2018-08-19:
```
int ledPin = 13;

void setup() {
  pinMode(ledPin, OUTPUT);
}

void loop() {
  digitalWrite(ledPin, HIGH);
  delay(200);
  digitalWrite(ledPin, LOW);
  delay(1000);
}
```

StaticRocket, 2019-01-05:
> I also see that it's using an older version of the Arduino, 1.6.21, right? Can that be an issue?

Ordinarily it could, but I just compiled this on 1.8.5 without any issues.