Sorunome/arduino-upload #60: Flex-bar buttons does not work...

StefanMeGit, 2018-11-11:
I installed everything and the flexbar works, but I can't use the buttons u described... when go over PACKAGES - ARDUNIO UPLOAD - VERIFY everything works as expected.

The just do nothing... other buttons like new-file working good. Any idea?
```
  {
    type: "button"
    icon: "file-add"
    tooltip: "New File"
    callback: "application:new-file"
  }
```

```
  {
    type: "button"
    icon: "check"
    callback: ["core:save", "arduino-upload:verify"]
    tooltip: "Arduino: Verify",
    enable: { grammar: "arduino" }
  }
  {
    type: "button"
    icon: "arrow-right"
    callback: ["core:save", "arduino-upload:upload"]
    tooltip: "Arduino: Upload",
    enable: { grammar: "arduino" }
  }
  {
    type: "button"
    icon: "terminal"
    callback: "arduino-upload:serial-monitor"
    tooltip: "Arduino: Serial monitor",
    enable: { grammar: "arduino" }
  }
  {
    type: "spacer"
  }
  {
    type: "button"
    icon: "gear"
    callback: "flex-tool-bar:edit-config-file"
    tooltip: "Edit Tool Bar"
  }
```

StaticRocket, 2018-11-14:
You forgot to separate each section and subsection with commas.
If you're using JS config, it should look like this:
```javascript
{
    type: "button",
    icon: "check",
    callback: ["core:save", "arduino-upload:verify"],
    tooltip: "Arduino: Verify",
    enable: { grammar: "arduino" }
  },
  {
    type: "button",
    icon: "arrow-right",
    callback: ["core:save", "arduino-upload:upload"],
    tooltip: "Arduino: Upload",
    enable: { grammar: "arduino" }
  },
  {
    type: "button",
    icon: "terminal",
    callback: "arduino-upload:serial-monitor",
    tooltip: "Arduino: Serial monitor",
    enable: { grammar: "arduino" }
  },
  {
    type: "spacer"
  },
  {
    type: "button",
    icon: "gear",
    callback: "flex-tool-bar:edit-config-file",
    tooltip: "Edit Tool Bar"
  }
```

---
Also, you don't need to use core save, I kind of made an option that can do this automatically on build/verification.

jrial, 2018-12-01:
In that case someone needs to fix the documentation at https://github.com/Sorunome/arduino-upload/blob/v1.4.1/docs/toolbar.md

I was already wondering why there were no commas in the cson.

closed 2019-01-05 by StaticRocket