Sorunome/arduino-upload #63: Build issues depending on selected board

jrial, 2018-12-01:
I've been having some issues with failed builds for a couple of weeks, and finally found a way to get them to work. The cause is strange though, and I can't explain what's going on. Reporting the issue since it appears there's some weirdness going on between Arduino-upload and the Arduino IDE.

I'm trying to compile a sketch for the NodeMCU board, and these are my results:

1. First build was in Atom with NodeMCU 0.9 selected. It failed:

[image]

2. Then I went into Arduino IDE, selected the NodeMCU 0.9 board and performed a successful build:

[image]

3. After that, the new build succeeded without a hitch:

[image]

I also noticed that if I set Atom to use the Arduino Settings for the board, and have the right board selected in the Arduino IDE, the build also succeeds. But unless I also perform a build in the Arduino IDE, I can not get the build to succeed under Atom when selecting the board directly.

StaticRocket, 2018-12-24:
@jrial I'm going to try to replicate this so, just to be sure, where are you getting this board's config from? Did you just add `http://arduino.esp8266.com/stable/package_esp8266com_index.json` to the Additional Boards Manager section in Arduino IDE's settings?

jrial, 2018-12-31:
It contains these two extra board URLs: `http://arduino.esp8266.com/stable/package_esp8266com_index.json,https://dl.espressif.com/dl/package_esp32_index.json`

StaticRocket, 2019-01-06:
Alright, that board config actually has its own options that are set inside the Arduino IDE through new entries in the Tools context menu. We currently don't support these advanced options (I didn't even know they existed until now), but you should be able to adjust them in the IDE and use the `== Arduino Setting ==` board entry or a custom entry with our package to work around this. Unfortunately these options are required for verifying, building and uploading the sketch as they all work on the same build method.

---

**Context menu entries for reference**

Normal:

[image]

With NodeMCU 0.9 (ESP-12 Module) board selected:

[image]