Sorunome/arduino-upload #65: Uncaught Error: spawn EPERM

Paqoz14, 2018-12-22:
Hey Guys, I'm new to this package and I don't know how to get this thing working...
When I try to verify the current sketch I get this error:

**Atom**: 1.33.1 x64
**Electron**: 2.0.16
**OS**: Windows 10 Pro
**Thrown From**: [arduino-upload](https://github.com/Sorunome/arduino-upload) package 1.4.1

### Stack Trace

Uncaught Error: spawn EPERM

```
At internal/child_process.js:323

Error: spawn EPERM
    at _errnoException (util.js:1024:11)
    at ChildProcess.spawn (internal/child_process.js:323:11)
    at exports.spawn (child_process.js:514:9)
    at Object._build (/packages/arduino-upload/lib/arduino-upload.coffee:178:15)
    at Object.build (/packages/arduino-upload/lib/arduino-upload.coffee:218:4)
    at /packages/arduino-upload/lib/arduino-upload.coffee:97:33)
    at CommandRegistry.handleCommandEvent (~/AppData/Local/atom/app-1.33.1/resources/app/static/<embedded>:11:350034)
    at CommandRegistry.dispatch (~/AppData/Local/atom/app-1.33.1/resources/app/static/<embedded>:11:348509)
    at ToolBarButtonView.executeCallback (/packages/tool-bar/lib/tool-bar-button-view.js:130:21)
    at ToolBarButtonView._onClick (/packages/tool-bar/lib/tool-bar-button-view.js:109:12)
```

### Commands

```
     -3:39.5.0 symbols-view:toggle-project-symbols (input.hidden-input)
 10x -3:36.6.0 core:backspace (input.hidden-input)
     -3:29.8.0 arduino-upload:verify (input.hidden-input)
  2x -3:06.4.0 arduino-upload:upload (input.hidden-input)
     -2:28.4.0 window:toggle-dev-tools (input.hidden-input)
     -2:17.9.0 arduino-upload:serial-monitor (input.hidden-input)
  4x -1:17.6.0 core:save (input.hidden-input)
     -1:03.3.0 arduino-upload:verify (input.hidden-input)
     -0:59.8.0 arduino-upload:build (input.hidden-input)
     -0:52 arduino-upload:upload (input.hidden-input)
     -0:35.7.0 arduino-upload:serial-monitor (input.hidden-input)
     -0:28.6.0 arduino-upload:upload (input.hidden-input)
     -0:17.6.0 grammar-selector:show (atom-text-editor.editor.is-focused)
  2x -0:12.5.0 core:save (input.hidden-input)
     -0:11.1.0 arduino-upload:upload (input.hidden-input)
     -0:05.1.0 arduino-upload:verify (input.hidden-input)
```

### Non-Core Packages

```
arduino-upload 1.4.1 
atom-beautify 0.33.4 
flex-tool-bar 2.1.2 
language-arduino 0.4.3 
local-history 4.3.1 
teletype 0.13.3 
tool-bar 1.1.11 
```

StaticRocket, 2018-12-24:
I can't seem to replicate this issue. Is there any more information you could give?

Paqoz14, 2018-12-24:
I've installed the arduino-upload package, because I don't like the normal IDE. When I'm trying to upload the code to the Arduino I get this error. I think it's because I don't have permission to access the folder, where the normal IDE is installed, but the package needs access to the files of the normal IDE. I tried to change the access Settings of the Folder, but it doesn't solve the problem. I hope, that it will work, when I install the normal IDE to an other Folder and not with the Microsoft Store. So I think I know the reason of the error and I'm working on it to get it fixed. I'm sorry for my bad English, I'm German.

Paqoz14, 2018-12-24:
Yeah, I'll close the issue. I think it's better if you write the note in README, my English is maybe to bad. Thanks!

Paqoz14, 2018-12-24:
I installed the IDE in the ProgramFiles Folder now and changed the settings of the package. It works now! Thanks!

closed 2018-12-24 by Paqoz14