Sorunome/arduino-upload #66: Build fails in Windows (bug in generated command line)

AriathTheSage, 2019-01-03:
Hi, i'm testing this with Atom in Windows 10 and, when i call "Arduino upload ->Build", i get these errors in the console:

> events.js:183 Uncaught Error: ENOENT: no such file or directory, open 'C:\Users\Javier\AppData\Local\Temp\tmp-9284pqoB2lUTpWqhArduinoTest.ino.eep'
events.js:183 Uncaught Error: ENOENT: no such file or directory, open 'C:\Users\Javier\AppData\Local\Temp\tmp-9284pqoB2lUTpWqhArduinoTest.ino.elf'
events.js:183 Uncaught Error: ENOENT: no such file or directory, open 'C:\Users\Javier\AppData\Local\Temp\tmp-9284pqoB2lUTpWqhArduinoTest.ino.hex'
events.js:183 Uncaught Error: ENOENT: no such file or directory, open 'C:\Users\Javier\AppData\Local\Temp\tmp-9284pqoB2lUTpWqhArduinoTest.ino.bin'

It seems the command line is malformed because a missing backslash between the temp folder name "tmp-9284pqoB2lUTpWqh" and the file.

Thanks for a good and simple Arduino plugin :3 .

StaticRocket, 2019-01-03:
Yeah... that's not cool. Just out of curiosity what does your project's directory tree look like? Did you generate those files or did the package?  (The mismatched file and directory names seem odd, but you should have gotten an error about that earlier if that is the case.)

AriathTheSage, 2019-01-03:
As simple as M:\Proyectos\Arduino\ArduinoTest\ArduinoTest.ino.
Created with the Arduino IDE.
Verify works ok, 0 errors.

After the build error, if you open the sketch folder, you can see the eep, elf, hex and bin files with 0 bytes size.

StaticRocket, 2019-01-03:
That doesn't make any sense. Everything seems fine and I can't replicate the error. Perhaps move the temp folder somewhere else for now and try rebuilding it? (I know. It's just a temporary fix if it even fixes it at all, but I'm curious.) If that works can you post a photo of the newly generated temp directory?

AriathTheSage, 2019-01-03:
I think MAYBE i have found the error (first time i use Atom, and first time i debug a plugin here :) ).

`build: (keep) ->
		@_build ['--verify'], (code, info) =>
			if code != false
				if code != 0
					atom.notifications.addError 'Build failed!'
				else if keep
					for ending in ['.eep', '.elf', '.hex']
						fs.createReadStream(info.buildFolder + seperator + info.name + ending).pipe(fs.createWriteStream(info.workpath + seperator + info.name + ending))`

I have put a "seperetor" (btw, i think it would be better to name it "separator") between info.buildFolder and info.name, in the first parameter of fs.createReadStream function.

Also, i have removed the ".bin" file extension because it seems no ".bin" file exists.

Now, it creates the hex file correctly (checked with WinMerge), but is the only file that copies :/ ...

StaticRocket, 2019-01-04:
Huh. That's like... a fundamental issue. How did the package even work before this? I guess `info.buildFolder` occasionally outputs a path followed by a separator, but in your case it didn't. Even still, how did this exist so long without throwing any errors for anyone else? @Sorunome am I just being dumb or is this odd.

---

Also you can paste a link directly to the code your referencing and it gives a nice preview to save time.

Ex: `https://github.com/Sorunome/arduino-upload/blob/fba9a5c94d1f4e55725f4523cd79652f564de53f/lib/arduino-upload.coffee#L217-L224` Produces this:
https://github.com/Sorunome/arduino-upload/blob/fba9a5c94d1f4e55725f4523cd79652f564de53f/lib/arduino-upload.coffee#L217-L224

AriathTheSage, 2019-01-04:
Thanks for the info, i'm quite newbie in GitHub :) .
Maybe info.buildFolder changed and now it doesn't include the path separator...

Btw, maybe it is better now because the folder separator is platform dependant and i think it shouldn't be included here :) .

StaticRocket, 2019-01-04:
I'm not getting any errors with or without your proposed changes on Arch Linux or Windows 10. Really odd. If I don't hear anything from Sorunome in a few hours I'll either patch it in as a precaution, or merge your PR if you want to submit one (since you managed to find and fix something I didn't even know was a problem).

AriathTheSage, 2019-01-05:
I prefer you patch it. I don't have GitHub installed nor have made ever a PR :S .

Sorunome, 2019-01-05:
The easy fix would OFC be to just add the `seperator` after `info.buildFolder`, as `//` and `\\` on linux and windows respectively are treated as if they were just one `/` and one `\`. The clean way would be to use some path library or something

closed 2019-01-05 by StaticRocket