Sorunome/arduino-upload #69: Unexpected behavior of Build, Upload and Verify functions

Giovimax, 2019-01-25:
Hi, I'm using Atom 1.34.0 on Ubuntu 18.10, arduino IDE version should be 2:1.0.5+dfsg2-4.1 .
The version of arduino-upload is 1.5.1 .
The first abnormal behavior is that on atom's start up, a blank window of Arduino IDE opens up, if its closed if freezes atom. 
Each time I run Verify or Build, a new Arduino IDE window pops up but nothing else happens. Wierdly, trying the Upload function, the uploard is successfull but the Arduino IDE's window shows an error (Bad file selected) due to the .ino extension not begin transmitted to it at the end of the file's name.
The Serial Monitor instead seems to don't have any problems and does not open up a new arduino IDE's window.
I have alredy set the IDE's language to English so I suppose the issue isn't related to that.
Thanks in advance fot the help.

Sorunome, 2019-01-25:
try updating your arduino IDE version, it needs to be at least 1.6.x

Giovimax, 2019-01-25:
Ok it worked: I uninstalled the previous version and installed version 1.8.8, now everything in working properly. 
Thank you a lot.

closed 2019-01-25 by Giovimax