Sorunome/arduino-upload #75: Failed to install on Atom 1.39

jgromes, 2019-07-26:
Hi,

After the latest Atom update, the package fails to install, this is the output when installing:

```
> @serialport/bindings@2.0.8 install C:\Users\User\AppData\Local\Temp\apm-install-dir-119626-6940-12oi6cw.avfg\node_modules\arduino-upload\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> usb-detection@4.3.0 install C:\Users\User\AppData\Local\Temp\apm-install-dir-119626-6940-12oi6cw.avfg\node_modules\arduino-upload\node_modules\usb-detection
> prebuild-install || node-gyp rebuild

C:\Users\User\AppData\Local\Temp\apm-install-dir-119626-6940-12oi6cw.avfg\node_modules\arduino-upload\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Users\User\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\User\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

prebuild-install WARN install No prebuilt binaries found (target=3.1.10 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Users\User\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:492:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Users\User\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:517:16)
gyp ERR! stack     at C:\Users\User\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Users\\User\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\User\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\User\AppData\Local\Temp\apm-install-dir-119626-6940-12oi6cw.avfg\node_modules\arduino-upload\node_modules\usb-detection
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\User\AppData\Local\Temp\apm-install-dir-119626-6940-12oi6cw.avfg\package.json'
npm WARN apm-install-dir-119626-6940-12oi6cw.avfg No description
npm WARN apm-install-dir-119626-6940-12oi6cw.avfg No repository field.
npm WARN apm-install-dir-119626-6940-12oi6cw.avfg No README data
npm WARN apm-install-dir-119626-6940-12oi6cw.avfg No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb-detection@4.3.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@4.3.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\User\.atom\.apm\_logs\2019-07-26T05_28_35_610Z-debug.log
```

Apparently, other packages have a similar problem: https://discourse.julialang.org/t/atom-1-39-wont-build-julia-client/26696/11

There's a workaround to install standalone Atom 1.38, but still, would it be possible to get this fixed? Thanks!

StaticRocket, 2019-07-27:
Yep, good old ELIFECYCLE. The `usb-detection` package needs to be rebuilt upstream. I wish there was a better solution to this but that package just seems to lag behind major updates.
`npm`
```
> usb-detection@4.3.0 install C:\Users\Res\Desktop\arduino-upload\node_modules\usb-detection
> prebuild-install \|\| node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=11.6.0 runtime=node arch=x64 libc= platform=win32)

C:\Users\Res\Desktop\arduino-upload\node_modules\usb-detection>if not defined npm_config_node_gyp (node "D:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-g yp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "D:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Command failed: C:\Users\Res\AppData\Local\Programs\Python\Python37\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];                        gyp ERR! stack   File "", line 1
gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
gyp ERR! stack
gyp ERR! stack SyntaxError: invalid syntax
gyp ERR! stack
gyp ERR! stack     at ChildProcess.exithandler (child_process.js:294:12)
gyp ERR! stack     at ChildProcess.emit (events.js:188:13)
gyp ERR! stack     at maybeClose (internal/child_process.js:978:16)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:265:5)
gyp ERR! System Windows_NT 10.0.17763
gyp ERR! command "D:\\Program Files\\nodejs\\node.exe" "D:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\Res\Desktop\arduino-upload\node_modules\usb-detection
gyp ERR! node -v v11.6.0 
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb-detection@4.3.0 install: `prebuild-install \|\| node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@4.3.0 install script. 
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.       
```

lorenzos, 2019-07-27:
Is there a workaround for this, in the meantime? Thanks.

jgromes, 2019-07-27:
> There's a workaround to install standalone Atom 1.38

Yes, get a standalone Atom 1.38 from https://github.com/atom/atom/releases/tag/v1.38.2
It should use all packages and custom configuration of the regular installation, you just can't update.

lorenzos, 2019-07-27:
Thanks, I'm on Ubuntu, so I downgraded quicly using the package manager. In case anyone will need it:

    sudo apt-get install atom=1.38.2

Rebooting Atom will ask to rebuild the Arduino upload package (I guess because I tried to rebuild previosly, when it gave the error), and then everything worked.

StaticRocket, 2019-07-27:
Closing because were at the mercy of the packager for this one.

https://github.com/MadLittleMods/node-usb-detection/issues/85

StaticRocket, 2019-08-13:
Absolutely jumped the gun on this. I really should leave it open for it's information.