Sorunome/arduino-upload #76: arduino-upload not installing

zhan965, 2019-08-04:
Hi, i have been using arduino-upload for the past 4 or 5 months without any problems untill today, it stopped uploading to my arduino anything, i tried to reinstall the package, but now it just gives me an error every time i try, please help, will attach error message bellow. specs: windows 10, latest version of atom.

Installing “arduino-upload@1.5.1” failed.Hide output…

> @serialport/bindings@2.0.8 install C:\Users\zhan9\AppData\Local\Temp\apm-install-dir-11974-15824-nhlnv3.b56s\node_modules\arduino-upload\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> usb-detection@4.3.0 install C:\Users\zhan9\AppData\Local\Temp\apm-install-dir-11974-15824-nhlnv3.b56s\node_modules\arduino-upload\node_modules\usb-detection
> prebuild-install || node-gyp rebuild

C:\Users\zhan9\AppData\Local\Temp\apm-install-dir-11974-15824-nhlnv3.b56s\node_modules\arduino-upload\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Users\zhan9\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\zhan9\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild ) 

prebuild-install WARN install No prebuilt binaries found (target=3.1.10 runtime=electron arch=x64 libc= platform=win32)
gyp ERR! configure error 
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at PythonFinder.failNoPython (C:\Users\zhan9\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:492:19)
gyp ERR! stack     at PythonFinder.<anonymous> (C:\Users\zhan9\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\node-gyp\lib\configure.js:517:16)
gyp ERR! stack     at C:\Users\zhan9\AppData\Local\atom\app-1.39.1\resources\app\apm\node_modules\npm\node_modules\graceful-fs\polyfills.js:284:29
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:152:21)
gyp ERR! System Windows_NT 10.0.17134
gyp ERR! command "C:\\Users\\zhan9\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\zhan9\\AppData\\Local\\atom\\app-1.39.1\\resources\\app\\apm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\zhan9\AppData\Local\Temp\apm-install-dir-11974-15824-nhlnv3.b56s\node_modules\arduino-upload\node_modules\usb-detection
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.7.0
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\zhan9\AppData\Local\Temp\apm-install-dir-11974-15824-nhlnv3.b56s\package.json'
npm WARN apm-install-dir-11974-15824-nhlnv3.b56s No description
npm WARN apm-install-dir-11974-15824-nhlnv3.b56s No repository field.
npm WARN apm-install-dir-11974-15824-nhlnv3.b56s No README data
npm WARN apm-install-dir-11974-15824-nhlnv3.b56s No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb-detection@4.3.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@4.3.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\zhan9\.atom\.apm\_logs\2019-08-04T12_24_06_187Z-debug.log

zhan965, 2019-08-11:
any help would be greatly appreciated, pls pls pls

Sorunome, 2019-08-12:
> gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable

sounds like you might need python for it to be able to compile one of the dependencies?

vivi90, 2020-10-29:
@zhan965 Does this issue still exists?