Sorunome/arduino-upload #83: Build error without information

alve89, 2020-01-25:
Is there an option to get the error messages why a build doesn't work? The only error message is 
> Build failed

Because without this information you need to compile it in the Arduino IDE to get the error contents...

geovani25, 2020-08-01:
Same thing happening here.

When I compile in Arduino IDE it compiles fine.

[image]

geovani25, 2020-08-01:
Same thing happening in here.
But when compiling in Arduino IDE it runs fine.

[image]