Sorunome/arduino-upload #90: Package installation causing Atom to crash.

LuvaiCutlerywala, 2021-04-15:
The installation of the package causes the Atom Editor to simply crash. When it happened the first time, I thought some configuration of mine caused the editor to crash, so I removed the install as well cleared all the configuration files. Then, after I reinstalled Atom from scratch, I downloaded the package again, and again the editor simply crashes every time I try to load the application.

Puzzlepack, 2021-04-21:
The same happened here. The only solutions was to reinstall Atom...

donaca4555, 2021-06-02:
> The installation of the package causes the Atom Editor to simply crash. When it happened the first time, I thought some configuration of mine caused the editor to crash, so I removed the install as [...]

same here. delete the package folder fix the crashes.

LuvaiCutlerywala, 2021-06-03:
Funny thing is it kept on crashing even after I did that. So any help as to why it happened or what I could do to remedy it would be appreciated.

donaca4555, 2021-06-07:
In my case I couldn't wait for a fix and went with platformIO plugin, it works great for my arduino project. (upload, monitor and more..)

lorenzos, 2021-06-07:
I tried the new official beta 2.0 IDE, but it's absolutely underwhelming compared to Atom (but still better than the legacy one, of course).

Did anyone tried to downgrade Atom? Could that work? 

@donaca4555 I'm tempted trying that way too. Does it work without signing up and with local (non-cloud) projects?

jackik1410, 2021-08-12:
Same issue here, none of the tools worked at any point, kept crashing atom once installed. Fix for the crashes was running `apm disable arduino-upload`.
Really wish this worked at the moment, would have saved me a lot of time.

lorenzos, 2021-08-12:
I ended up using Arduino CLI and implementing some toolbar command launched in a terminal plugin. Not bad: Arduino CLI is a very nice piece of software, my Atom integration is a little cumbersome, but I'm happy with it. 

If someone is interested, I will spend some minutes to share it.

LuvaiCutlerywala, 2021-08-12:
I’d love to know the setup, as mine currently is just the beta IDE, which as described before is underwhelming.

lorenzos, 2021-08-12:
So, here's my setup. First thing, I [installed Arduino CLI](https://arduino.github.io/arduino-cli/latest/installation/) and followed the [Getting started guide](https://arduino.github.io/arduino-cli/latest/getting-started/) to make sure it runs fine from the command line.

Then, these are the Atom packages to install for toolbar and terminal:

```txt
flex-tool-bar
language-arduino
terminal-commands
tool-bar
x-terminal
```

This is the configuration for the terminal package:

```cson
  "x-terminal":
    spawnPtySettings:
      projectCwd: true
    terminalSettings:
      colors:
        selection: "#ffffff"
      defaultOpenPosition: "Split Down"
      leaveOpenAfterExit: false
      runInActive: true
      title: "Terminal"
    xtermAddons: {}
```

This is the `terminal-commands.json` file:

```json
{
  "arduino-cli:compile": [
      "direnv allow \"${project}\"", 
      "arduino-cli compile -b \"$ARDUINO_FQBN\" \"${dir}\""
  ],
  "arduino-cli:upload": [
      "direnv allow \"${project}\"", 
      "arduino-cli compile -b \"$ARDUINO_FQBN\" -u -p \"$ARDUINO_PORT\" \"${dir}\" && echo \"Done uploading.\""
  ],
  "arduino-cli:board-list": [
      "test -f \"${project}/.envrc\" && atom \"${project}/.envrc\"", 
      "arduino-cli board list"
  ],
  "arduino-cli:serial-monitor": [
      "screen $ARDUINO_PORT"
  ]
}
```

I use [`direnv`](https://direnv.net/) to store and change `$ARDUINO_FQBN` and `$ARDUINO_PORT` at will. It's optional, since those variable can be replaced by actual fixed values in `terminal-commands.json` (if so, remove all `direnv allow` commands` from it). If used, here's an example `.envrc` file:

```
export ARDUINO_FQBN=arduino:avr:nano:cpu=atmega328old
export ARDUINO_PORT=/dev/ttyUSB0
```

Finally, here's the `toolbar.cson`:

```cson
[
  {
    type: "button"
    icon: "check"
    callback: ["window:save-all", "arduino-cli:compile"]
    tooltip: "Arduino: Save all and Compile"
  }
  {
    type: "button"
    icon: "arrow-right"
    callback: ["window:save-all", "arduino-cli:upload"]
    tooltip: "Arduino: Save all and Upload"
  }
  {
    type: "button"
    icon: "terminal"
    callback: "arduino-cli:serial-monitor"
    tooltip: "Arduino: Serial monitor"
  }
  {
    type: "button"
    icon: "plug"
    callback: "arduino-cli:board-list"
    tooltip: "Arduino: List connected boards"
  }
  {
    type: "spacer"
  }
  {
    type: "url"
    icon: "circuit-board"
    url: "https://www.arduino.cc/reference/en/"
    tooltip: "Open: Arduino reference"
  }
  {
    type: "spacer"
  }
  {
    type: "button"
    icon: "tools"
    callback: "settings-view:view-installed-packages"
    tooltip: "Atom packages settings"
  }
  {
    type: "button"
    icon: "gear"
    callback: "flex-tool-bar:edit-config-file"
    tooltip: "Edit Toolbar"
  }
]
```

LuvaiCutlerywala, 2021-08-13:
Sounds good. Thanks for giving the setup, I’ll try it when I get the chance to do so.

steeley, 2022-01-23:
Got the same result - totally trashed my entire Atom set up!!!!!!!!!!
Thanks for that....

Never had a problem with other Atom extensions.

maccacino, 2022-02-04:
still crashing...
For Ubuntu (and most Linux systems) run this command in a terminal;

`apm uninstall arduino-upload`

That is using Atom's package manager to remove the broken package, without having to re-install Atom.
[Atoms Flight Manual](https://flight-manual.atom.io/using-atom/sections/atom-packages/#command-line)

CarterSnich, 2022-03-27:
> So, here's my setup.
> 
> First thing, I [installed Arduino CLI](https://arduino.github.io/arduino-cli/latest/installation/) and followed the [getting started guide](https://arduino.github.io/arduino-cli/latest/getting-started/) to make sure it runs fine from the command line.
> 
> Then, these are the Atom packages to install for toolbar and terminal:
> 
> ```
> flex-tool-bar
> language-arduino
> terminal-commands
> tool-bar
> x-terminal
> ```
> 
> This is the configuration for the terminal package:
> 
> ```coffee
>   "x-terminal":
>     spawnPtySettings:
>       projectCwd: true
>     terminalSettings:
>       colors:
>         selection: "#ffffff"
>       defaultOpenPosition: "Split Down"
>       leaveOpenAfterExit: false
>       runInActive: true
>       title: "Terminal"
>     xtermAddons: {}
> ```
> 
> This is the `terminal-commands.json` file:
> 
> ```json
> {
>   "arduino-cli:compile": [
>       "direnv allow \"${project}\"", 
>       "arduino-cli compile -b \"$ARDUINO_FQBN\" \"${dir}\""
>   ],
>   "arduino-cli:upload": [
>       "direnv allow \"${project}\"", 
>       "arduino-cli compile -b \"$ARDUINO_FQBN\" -u -p \"$ARDUINO_PORT\" \"${dir}\" && echo \"Done uploading.\""
>   ],
>   "arduino-cli:board-list": [
>       "test -f \"${project}/.envrc\" && atom \"${project}/.envrc\"", 
>       "arduino-cli board list"
>   ],
>   "arduino-cli:serial-monitor": [
>       "screen $ARDUINO_PORT"
>   ]
> }
> ```
> 
> I use [`direnv`](https://direnv.net/) to store and change `$ARDUINO_FQBN` and `$ARDUINO_PORT` at will. It's optional, since those variable can be replaced by actual fixed values in `terminal-commands.json` (if so, remove all `direnv allow` commands from it). If used, here's an example `.envrc` file:
> 
> ```
> export ARDUINO_FQBN=arduino:avr:nano
> export ARDUINO_PORT=/dev/ttyUSB0
> ```
> 
> Finally, here's the `toolbar.cson`:
> 
> ```coffee
> [
>   {
>     type: "button"
>     icon: "check"
>     callback: ["window:save-all", "arduino-cli:compile"]
>     tooltip: "Arduino: Save all and Compile"
>   }
>   {
>     type: "button"
>     icon: "arrow-right"
>     callback: ["window:save-all", "arduino-cli:upload"]
>     tooltip: "Arduino: Save all and Upload"
>   }
>   {
>     type: "button"
>     icon: "terminal"
>     callback: "arduino-cli:serial-monitor"
>     tooltip: "Arduino: Serial monitor"
>   }
>   {
>     type: "button"
>     icon: "plug"
>     callback: "arduino-cli:board-list"
>     tooltip: "Arduino: List connected boards"
>   }
>   {
>     type: "spacer"
>   }
>   {
>     type: "url"
>     icon: "circuit-board"
>     url: "https://www.arduino.cc/reference/en/"
>     tooltip: "Open: Arduino reference"
>   }
>   {
>     type: "spacer"
>   }
>   {
>     type: "button"
>     icon: "tools"
>     callback: "settings-view:view-installed-packages"
>     tooltip: "Atom packages settings"
>   }
>   {
>     type: "button"
>     icon: "gear"
>     callback: "flex-tool-bar:edit-config-file"
>     tooltip: "Edit Toolbar"
>   }
> ]
> ```

Is it worth creating a package that uses `android-cli`? I'm planning of creating one.

lorenzos, 2022-03-28:
> Is it worth creating a package that uses `android-cli`? I'm planning of creating one.

I don't know :man_shrugging: I still think it's better fixing this issue. My setup works, but IMHO it has too many dependencies to make a package out of it (not only `arduino-cli`, but also multiple toolbar and terminal packages); it also has some annoyances I found later, such as: it won't work if you launch the commands while the terminal tab has focus, instead of the editor tab with your sketch...

But, if you're thinking of creating a package that relies on `arduino-cli` ***only***, that implements all the rest itself, yeah, that would be definitely great. `arduino-cli` works great, it's well engineered, and I think it's in fact the officially suggested approach: as they say, it *"contains all you need to easily build applications around the Arduino ecosystem"*.